unosquare/raspberryio #5: Add Bluetooth Singleton

mariodivece, 2016-12-16:
Enable bluetooth manipulation via the Pi API

hidegh, 2018-02-18:
Just a question: why stick with singleton? Worst pattern to exist, especially with dependency injection. And on low level nothing prevents the user to create a global variable and or create custom static implementation with all the wiring needed - and store the result in a global variable...

I do understand that programming against one GPIO makes this static/singleton sound reasonable, but using static/singleton everywhere might limit possibilities (avail. via inheritance) in the long run.

geoperez, 2018-03-05:
@hidegh the idea behind the singleton is just to keep a single entry point for that particular API. The API is not completed, but you can see something here:

https://github.com/unosquare/raspberryio/blob/Bluetooth/src/Unosquare.RaspberryIO/Bluetooth/BluetoothController.cs

closed 2019-02-04 by stale[bot]