unosquare/raspberryio #14: .Net Core 2.0 on Win 10 IoT Supported?

PeteCozens, 2017-09-23:
I'm writing a .Net Core 2.0 Console App (not UWP) to control some external devices on a Raspberry Pi 3. Does RaspberryIO support this?

I've added the NuGet package  to my console app and added the following code to void main() to turn on a LED on pin 5:

var pin = Pi.Gpio.Pin05;
pin.PinMode = GpioPinDriveMode.Output;
pin.Write(true);

When stepping over the first line I get a System.TypeInitializationException (The type initializer for 'Unosquare.RaspberryIO.Pi' threw an exception)

System.TypeInitializationException: The type initializer for 'Unosquare.Raspberr
yIO.Pi' threw an exception. ---> System.Reflection.TargetInvocationException: Ex
ception has been thrown by the target of an invocation. ---> System.PlatformNotS
upportedException: This library does not support the platform
   at Unosquare.RaspberryIO.Gpio.GpioController.Initialize(ControllerMode mode)
   at Unosquare.RaspberryIO.Gpio.GpioController..ctor()
   --- End of inner exception stack trace ---
   at System.RuntimeTypeHandle.CreateInstance(RuntimeType type, Boolean publicOn
ly, Boolean& canBeCached, RuntimeMethodHandleInternal& ctor)
   at System.RuntimeType.CreateInstanceSlow(Boolean publicOnly, Boolean skipChec
kThis, Boolean fillCache, StackCrawlMark& stackMark)
   at System.RuntimeType.CreateInstanceDefaultCtor(Boolean publicOnly, Boolean s
kipCheckThis, Boolean fillCache, StackCrawlMark& stackMark)
   at System.Activator.CreateInstance(Type type, Boolean nonPublic)
   at Unosquare.Swan.Abstractions.SingletonBase`1.<>c.<.cctor>b__7_0()
   at System.Lazy`1.ViaFactory(LazyThreadSafetyMode mode)
   at System.Lazy`1.ExecutionAndPublication(LazyHelper executionAndPublication,
Boolean useDefaultConstructor)
   at System.Lazy`1.CreateValue()
   at Unosquare.Swan.Abstractions.SingletonBase`1.get_Instance()
   at Unosquare.RaspberryIO.Pi..cctor()
   --- End of inner exception stack trace ---
   at Unosquare.RaspberryIO.Pi.get_Gpio()
   at Echo.Program.SetLightState(Boolean state) in C:\Software\ParkSmart\Echo\Ec
ho\Program.cs:line 34

geoperez, 2017-09-25:
Hi, this library is not supported by Windows IoT since we rely on WiringPi and Linux device files.

The support to Win IoT is not yet on schedule.

closed 2017-09-25 by geoperez

tylerl0706, 2018-04-13:
Just wanted to chime in here and say that the PowerShell team is using this package in:

https://github.com/PowerShell/PowerShell-IoT

😄 

It's been awesome so far for our Raspbian PowerShell Core users. It'd be really cool if raspberryio could support Win IoT.