unosquare/raspberryio #25: Shutdown|Restart

FrankDoersam, 2018-01-03:
Hey,
i would miss an Interface to restart and shutdown the device.
Could you implement that in this fantastic library :)

Best regards
Frank

mariodivece, 2018-01-04:
Why not use the built-in SWAN ProcessRunner? just do something like:

```csharp
var shutdownResult = await ProcessRunner.GetProcessResultAsync("halt");
var restartResult = await ProcessRunner.GetProcessResultAsync("reboot");
```

It would be certainly nice to implement these methods as:
```csharp
Pi.Reboot();
Pi.Shutdown();
```

https://github.com/unosquare/swan#the-processrunner-class
https://unosquare.github.io/swan/api/Unosquare.Swan.Components.ProcessRunner.html

mariodivece, 2018-01-04:
This is now implemented in the latest commit. Reopen if the methods have issues or don't work.

closed 2018-01-04 by mariodivece

FrankDoersam, 2018-01-04:
Thank you for the fast and courteous support. The team is great!