unosquare/raspberryio #29: Callback executed multiple times and upon exit a wait for a callback prevents app from quitting...

hidegh, 2018-02-18:
I've created a sample project for ELECROW shield. 2 leds (one blinking) second turned on/off by button pressing down. It's a .NET standard 2.0 library running on Raspbian (sudo is needed)...

Now there are 2 issues:

1. one is that the button press event (even if it should react on one edge, even if it's hold down for a longer period) does execute the **callback multiple times** (sometimes) and multiple is from (2 to 8 times approx).

2. second: after **releasing handles** (maybe I do wrong) the **app. does not exists until I don't push the button** (so there's a wait that does not get cancelled)...

```
using System;
using System.Threading;
using System.Threading.Tasks;
using Unosquare.RaspberryIO;
using Unosquare.RaspberryIO.Gpio;

namespace LibRIO
{
    public class PiCode
    {
        public void Run()
        {
            //
            // Blinker
            var ledGreen = Pi.Gpio.Pin27;
            ledGreen.PinMode = GpioPinDriveMode.Output;

            var blinkerCancellationTokenSource = new CancellationTokenSource();
            var blinkerCancellationToken = blinkerCancellationTokenSource.Token;

            var taskBlinker = new Task(
                () =>
                {
                    Console.WriteLine("Starting blinker");

                    do
                    {
                        var value = ledGreen.Read();
                        value = !value;
                        ledGreen.Write(value);
                        Thread.Sleep(1000);

                    } while (!blinkerCancellationToken.IsCancellationRequested);

                },
                blinkerCancellationToken
            );

            //
            // Button - led
            var ledRed = Pi.Gpio.Pin28;
            ledRed.PinMode = GpioPinDriveMode.Output;

            var buttonRed = Pi.Gpio.Pin02;
            buttonRed.PinMode = GpioPinDriveMode.Input;
            buttonRed.InputPullMode = GpioPinResistorPullMode.PullUp;
            buttonRed.RegisterInterruptCallback(EdgeDetection.RisingEdge, () =>
            {
                var now = DateTime.Now;
                Console.WriteLine($"Button callback {now:HH:mm:ss.fffff}");
                var level = ledRed.Read();
                level = !level;
                ledRed.Write(level);
            });

            //
            // Main
            try
            {
                Console.WriteLine("Starting GPIO program...");

                taskBlinker.Start();

                Console.WriteLine("Press ENTER to exit...");
                Console.ReadLine();
            }
            finally
            {
                // Stop threads
                Console.WriteLine("Cancelling blinking...");
                blinkerCancellationTokenSource.Cancel();

                Console.WriteLine("Disposing GPIO...");
                Pi.Gpio.Dispose();

                Console.WriteLine("All OK to QUIT...");
            }

        }
    }
}

```

techyian, 2018-06-02:
Hi, any update on this?

Thanks.

geoperez, 2018-06-02:
Not really, we have been busy with other projects. But pull requests are always welcome.

closed 2019-02-04 by stale[bot]