unosquare/raspberryio #34: Add Component: Button

robertsundstrom, 2018-03-17:
I have a component that could be something to consider for Peripherals:

```csharp
public class Button
    {
        private GpioPin gpioPin;
        private ulong last_interrupt_time;

        public Button(GpioPin gpioPin)
        {
            this.gpioPin = gpioPin;

            Initialize();
        }

        private void Initialize()
        {
            gpioPin.InputPullMode = GpioPinResistorPullMode.PullDown;
            gpioPin.PinMode = GpioPinDriveMode.Input;
            gpioPin.RegisterInterruptCallback(EdgeDetection.RisingAndFallingEdges, OnClick);
        }

        private void OnClick()
        {
            if (!gpioPin.Read())
            {
                ulong interrupt_time = WiringPi.millis();
                if (interrupt_time - last_interrupt_time > 500)
                {
                    last_interrupt_time = interrupt_time;
                    Click?.Invoke(this, new EventArgs());
                }
            }
        }

        public event EventHandler<EventArgs> Click;
    }
```

It could be further improved, and rather than the unsuitable “Click” event there would be “Pressed” and “Released”.

robertsundstrom, 2018-03-18:
My improved Button class:

```csharp
using System;
using System.Linq;
using System.Threading.Tasks;
using Unosquare.RaspberryIO.Gpio;
using Unosquare.RaspberryIO.Native;

namespace RelaysTest
{
    public class Button
    {
        private GpioPin gpioPin;
        private ulong pressed_last_interrupt;
        private ulong released_last_interrupt;

        public Button(GpioPin gpioPin)
        {
            this.gpioPin = gpioPin;

            Initialize();
        }

        private void Initialize()
        {
            gpioPin.InputPullMode = GpioPinResistorPullMode.PullDown;
            gpioPin.PinMode = GpioPinDriveMode.Input;
            gpioPin.RegisterInterruptCallback(EdgeDetection.RisingAndFallingEdges, handleInterrupt);
        }

        private void handleInterrupt()
        {
            if (gpioPin.Read())
            {
                handleButtonPressed();
            }
            else
            {
                handleButtonReleased();
            }
        }

        private void handleButtonPressed()
        {
            ulong interrupt_time = WiringPi.millis();

            if (interrupt_time - pressed_last_interrupt > 500)
            {
                pressed_last_interrupt = interrupt_time;
                Pressed?.Invoke(this, new EventArgs());
            }
        }

        private void handleButtonReleased()
        {
            ulong interrupt_time = WiringPi.millis();

            if (interrupt_time - released_last_interrupt > 500)
            {
                released_last_interrupt = interrupt_time;
                Released?.Invoke(this, new EventArgs());
            }
        }

        public event EventHandler<EventArgs> Pressed;

        public event EventHandler<EventArgs> Released;
    }
}
```

Test:

```csharp
    public static class Extensions
    {
        public static GpioPin GetGpioPinByBcmPinNumber(this GpioController controller, int bcmPinNumber) =>
            controller.First(pin => pin.BcmPinNumber == bcmPinNumber);
    }

    class Program
    {
        static async Task Main(string[] args)
        {
            var relayChannel1 = GpioController.Instance.GetGpioPinByBcmPinNumber(17);
            relayChannel1.PinMode = GpioPinDriveMode.Output;

            var buttonPin = GpioController.Instance.GetGpioPinByBcmPinNumber(26);
            buttonPin.PinMode = GpioPinDriveMode.Input;

            var button = new Button(buttonPin);
            button.Pressed += (s, e) =>
            {
                Console.WriteLine("Pressed");
            };

            button.Released += (s, e) =>
            {
                Console.WriteLine("Released");
                var value = relayChannel1.Read();
                relayChannel1.Write(!value);
            };

            Console.ReadKey();
        }
    }
```

geoperez, 2018-03-20:
I'm going to integrate it

geoperez, 2018-03-20:
https://github.com/unosquare/raspberryio/commit/f2f546438b7e674964c3f0a928ac6cbefefd6e0b

closed 2018-03-28 by geoperez

buda2570, 2020-11-15:
Hi . I am using such a code, I have Pin noise problem and only button is fired, any solution?