unosquare/raspberryio #36: i2c add device typeinitialization exception

pmeloy, 2018-03-23:
Making my first attempt using this library and it's not going well. I don't know if its because of the lack of examples or if something is actually wrong.

Using VS2015 with Raspberry Pi3 and Raspbian Stretch with wiringpi enabled and RaspberryIO installed by Nuget two days ago.
My code:
`
using Unosquare.RaspberryIO;
.....
	class BMP180
	{
		public BMP180(int address)
		{
			try
			{
				var BMP180Device = Pi.I2C.AddDevice(address);
			}
			catch (Exception ex)
			{
				Console.WriteLine("Failed to register BMP180 at {0}. {1}", string.Format("{0:X2}", address), ex.GetType().FullName);
			}
		}
	}
`
And I get "Failed to register BMP180 at 77. System.TypeInitializationException"

I've been searching through all the documentation I can find and don't see anything else I should be doing.

geoperez, 2018-03-23:
Can you post the stacktrace?

pmeloy, 2018-03-23:
System.TypeInitializationException: The type initializer for 'Unosquare.RaspberryIO.Pi' threw an exception. ---> System.TypeLoadException: Could not resolve type with token 01000043 (from typeref, class/assembly Unosquare.Swan.ReflectionExtensions, Unosquare.Swan, Version=0.18.0.0, Culture=neutral, PublicKeyToken=null)
  at Unosquare.RaspberryIO.Pi..cctor () [0x0002e] in <24c16b38f03d4d8384ee43863d670a67>:0 
   --- End of inner exception stack trace ---
  at NexPi.BMP180..ctor (System.Int32 address) [0x00008] in <a4cc2f9a96f24eadb7b9a58e3f543101>:0 
  at NexPi.FrmMain.FrmMain_Load (System.Object sender, System.EventArgs e) [0x0006b] in <a4cc2f9a96f24eadb7b9a58e3f543101>:0 
  at System.Windows.Forms.Form.OnLoad (System.EventArgs e) [0x0001f] in <b0eb38f42f1041219336663febc755f3>:0 
  at System.Windows.Forms.Form.OnLoadInternal (System.EventArgs e) [0x00023] in <b0eb38f42f1041219336663febc755f3>:0

pmeloy, 2018-03-23:
This would probably be self evident if I could run/debug from VS rather than build from VS then run from Pi.

geoperez, 2018-03-23:
What version are you using? If you are using Unosquare.Raspberry.IO 0.13.0 please downgrade SWAN to 0.16.0.

pmeloy, 2018-03-23:
Nuget on the RPi doesn't seem to have a way to list installed packages or a downgrade option or even an uninstall. Tried from VS and it seems to be doing things but it's still on the RPi.

Can't believe they'd have an RPi Nuget with no way to list or remove packages. Guess its time to wipe the SD card and start from scratch.

pmeloy, 2018-03-23:
No good. It is IO 0.13 and Swan 0.16 but still the same error.

geoperez, 2018-04-02:
Hello,

Try to update your nuget to version 0.14.0, please.

geoperez, 2018-04-17:
@pmeloy ??

closed 2018-10-05 by geoperez