unosquare/raspberryio #38: No call whilst capturing video

pierslawson, 2018-03-26:
I'm trying out your library on a RPi B. Communicating with the GPIO pins seems to work great. I could also capture an image fine. However when I was trying to capture video, nothing was captured. If I set CaptureDisplayPreview = true, the preview window does not open. I wrote a little python script to capture video that seemed to work fine, so I'm fairly sure the camera is working.

Do you have any suggestions for debugging the problem? 

I don't suppose you have a way of actually attaching a debugger remotely to the .Net code?

geoperez, 2018-04-02:
I noticed the issue too, let me see what's going on.

geoperez, 2018-04-06:
Playground output 

```
pi@raspberrypi:~/rpio-playgound $ sudo dotnet Unosquare.RaspberryIO.Playground.dll
 15:02:11.269 INF >> Starting program at 06/04/2018 15:02:10
 15:02:12.415 USR << Press any key to stop reading the video stream . . .
 15:02:22.913 INF >> Capture Stopped. Received 0.000 Mbytes in 0 callbacks in 10.498 seconds
 15:02:26.043 INF >> Program finished.
```

pierslawson, 2018-04-08:
A minor fix to your code above, it should be a ReadLine after the "Press any key..." line:
```
   "Press any key to stop reading the video stream . . .".Info();
   Console.ReadLine();
```
I used your code and started receiving call backs. However, a second run of the program didn't get callbacks. Perhaps the code does not release the camera correctly? If I reboot the RPi, then the first run works, but subsequent runs fail to get callbacks. Any ideas?

geoperez, 2018-04-09:
Internally we call `raspivid` then when the `CloseVideoStream` is called we interrupt the `raspivid` process.

I was not able to reproduce the issue. After pressing any key the raspivid process is no longer there. It took a little bit to close.

pierslawson, 2018-04-10:
How long did you wait for raspivid to close? For me it was still running about ten minutes later. I've attached a screenshot of the htop output which shows raspivid is still running waiting on an event.
[image]

pierslawson, 2018-04-10:
Sorry closed it by mistake... still a problem for me.

pierslawson, 2018-04-10:
If I run htop as super user I can use it to send a SIGINT to the raspivid process and then it shuts down. Is it perhaps a permission thing could be stopping the ```ClosevideoStream``` from working... though I'm running my application using:
```
sudo ./MyConsoleApp
```

geoperez, 2018-04-10:
My raspi is just taking some seconds to close raspivid. If you run raspivid
manually it has the same behavior?

pierslawson, 2018-04-10:
If I just run `raspivid -o test.mpg -n` I see it appear in htop then disappear... but then I presume it is just closing itself as its 5 second default timeout is reached. The application sets an infinite timeout, so a different mechanism is being used to close the application.

geoperez, 2018-04-10:
You need to output to the console. Something like this:

`raspivid -o - -t 0 -w 1920 -h 1080 -n -ex night -vf -pf main -b 0 -fps 25 -g 25 -qp 23 -ih`

pierslawson, 2018-04-10:
So if I run your command I see a stream of garbage in the terminal (which I guess is to be expected) then pressing CTRL+C aborts the program. Whilst the program is running I see the entries in htop and when I have pressed CTRL+C they disappear from htop. 

Is that the test you wanted me to do?

geoperez, 2018-04-10:
Yes. It's odd, but this exactly that RaspberryIO is doing internally.

geoperez, 2018-04-17:
@Israel-94 can you create a new branch and return the CT to OpenVideoStream and test.

forki, 2018-04-18:
is there a more complete sample that shows how to save the video or transfer it somewhere useful?

forki, 2018-04-18:
Yes was just hoping you could expand it to show how to save it properly or how to stream it somewhere properly. Current it's just a buffer and I'm not sure what I want to do next.

geoperez, 2018-04-18:
You can use FileStream to open a file and then call the Write method (
https://msdn.microsoft.com/en-us/library/system.io.filestream.write(v=vs.110).aspx
)

mariodivece, 2018-04-19:
Maybe we should kill the `raspivid` process by sending a Unix Signal?
https://github.com/unosquare/pigpio-dotnet/blob/master/Unosquare.PiGpio/NativeMethods/Utilities.cs#L217

RaiseSignal(2) is equivalent to Ctrl-C

Israel-94, 2018-04-24:
@pierslawson Could you check the [branch](https://github.com/unosquare/raspberryio/tree/Issue38-CTToOpenVideoStream) and tell us if the problem is resolve

pierslawson, 2018-04-26:
I took a copy of that branch and built it locally, then referenced the local copy of the project from my own project (instead of using the nuget package). I still get the same problem. 

Can I check... the fix you have tried does not require me to change any of my code does it? As far as i can see the only difference is that the Unosquare.Swan.Lite.dll is now 0.29. The Unosquare.Swan.dll is still 0.28.1. Is that correct?

Israel-94, 2018-04-26:
@pierslawson I added the cancellation token to the `OpenVideoStream()` task, so when you call the ` Pi.Camera.CloseVideoStream()` it should now stop the task and kill the `raspivid `process. Could you paste the code where you use the PI.Camera to check it out?

pierslawson, 2018-04-29:
Here it is:
```
        private static void TestCaptureVideo()
        {
            // Setup our working variables
            var videoByteCount = 0;
            var videoEventCount = 0;
            var startTime = DateTime.UtcNow;

            // Configure video settings
            var videoSettings = new CameraVideoSettings()
            {
                CaptureTimeoutMilliseconds = 0,
                CaptureDisplayPreview = false,
                ImageFlipVertically = true,
                CaptureExposure = CameraExposureMode.Night,
                CaptureWidth = 1920,
                CaptureHeight = 1080
            };

            try
            {
                // Start the video recording
                Pi.Camera.OpenVideoStream(videoSettings,
                    onDataCallback: (data) =>
                    {
                        videoByteCount += data.Length;
                        videoEventCount++;
                    },
                    onExitCallback: null);

                // Wait for user interaction
                startTime = DateTime.UtcNow;
                //Console.WriteLine("Press any key to stop reading the video stream . . .");
                //Console.ReadLine();
                Thread.Sleep(3000);
            }
            catch (Exception ex)
            {
                Console.WriteLine($"{ex.GetType()}: {ex.Message}");
            }
            finally
            {
                // Always close the video stream to ensure raspivid quits
                Pi.Camera.CloseVideoStream();

                // Output the stats
                var megaBytesReceived = (videoByteCount / (1024f * 1024f)).ToString("0.000");
                var recordedSeconds = DateTime.UtcNow.Subtract(startTime).TotalSeconds.ToString("0.000");
                Console.WriteLine($"Capture Stopped. Received {megaBytesReceived} Mbytes in {videoEventCount} callbacks in {recordedSeconds} seconds");
            }
        }
```

Israel-94, 2018-05-03:
@pierslawson @ianByrne Made some changes, the `raspivid ` should be killed after the `CloseVideoStream` is called, and the callback when reopening the `VideoStreamOpen` should receive data now.

ianByrne, 2018-05-04:
Thanks @Israel-94 , this seems to now resolve my issue.

geoperez, 2018-05-04:
@pierslawson can you check it?

pierslawson, 2018-05-04:
raspivid is still executing after my application exits. If I put a `Thread.Sleep(1000);` before my program exits, raspivid is closed down correctly and I run my program again.

It looks to me like the cancellation needs to wait for the signal to have been sent rather than immediately returning.

Israel-94, 2018-05-08:
@pierslawson, I made a change to the video task, now It's going to be waited to finish and kill the `raspivid` process, Could you test to see if the issue is resolved?

pierslawson, 2018-05-09:
At first I didn't think it was working because I wasn't seeing the raspivid process starting at all. But when I increased the time it was recording from 3 seconds to 10 seconds I did see data starting to be captured... and when I ran for 20 seconds I saw the raspivid running in htop... and closing down correctly.

So you appear to have solved the shutdown issue, though it now appears to take longer for the process to start up. Before it was recording data within the three seconds... now it is taking consistently longer. I'm not sure if you mind a longer start-up time?

geoperez, 2018-05-09:
How much longer? @Israel-94 did you notice this behavior?

Israel-94, 2018-05-09:
@pierslawson @geoperez No, I did not, that's not happening on our side
Here are some tests, and the raspivid process open and closes well

[image]

[image]

pierslawson, 2018-05-09:
I will try to revert the code and find out how long it used to take.

pierslawson, 2018-05-09:
I reverted the code... when using the older version of swan (0.29) the time to first capture was always 2.2 seconds. With your latest code (which uses swan 0.30) the time to first capture was always 3.2 seconds.

pierslawson, 2018-05-09:
When not running under a debugger, time to first capture (using swan 0.29) is 1.8 seconds.

pierslawson, 2018-05-10:
If I take the latest code in branch Issue38-CTToOpenVideoStream and change it to reference swan 0.29 (rather than swan 0.30) my code speeds up again, taking 2.2 seconds to get the first data back from raspivid (under the debugger). However, that also reintroduces the issue whereby raspivid does not shut down.

Israel-94, 2018-05-10:
@pierslawson The swan updated should not affect the raspberryio code, it just changes two lines to correct use one method, The change that really matters is the one that waits to the task to finish and kill the `raspivid` process and then continue the execution or stop the program

geoperez, 2018-05-17:
Ok let me perform some testing on my end.

ianByrne, 2018-05-21:
@geoperez @Israel-94 do you have any ETA of when these fixes will be pulled into an official release? Particularly the one with raspivid not restarting after stopping/starting the stream.

geoperez, 2018-05-21:
Not really, I need some time to reproduce the same steps described by @pierslawson 

Do you have any similar issue with the timing?

ianByrne, 2018-05-21:
I have not experienced any noticeable changes in timing between master and the branch, but I'm afraid I have not been looking out for it either.

Israel-94, 2018-06-26:
New release of [RaspberryIO](https://www.nuget.org/packages/Unosquare.Raspberry.IO/), closing issue

closed 2018-06-26 by Israel-94