unosquare/raspberryio #42: >.net standard class library

RunnicFusion, 2018-04-15:
It's not possible to use the Raspberry.io inside .net standard 2.0. Nuget package will install but gives errors when debugging.

How to fix this?

RunnicFusion, 2018-04-16:
> Hi,
>
> That the targeting framework is not supported (.net standard)
>
>
> Thanks for your reply!
>
>
>
> Met vriendelijke groet,
>
> Roy Wagemakers

.net core says:

System.TypeInitializationException: The type initializer for
'Unosquare.RaspberryIO.Pi' threw an exception. --->
System.Reflection.TargetInvocationException: Exception has been thrown by
the target of an invocation. --->
System.PlatformNotSupportedException: This library does not support the
platform
   at Unosquare.RaspberryIO.Gpio.GpioController.Initialize(ControllerMode
mode)
   at Unosquare.RaspberryIO.Gpio.GpioController..ctor()
   --- End of inner exception stack trace ---
   at System.RuntimeTypeHandle.CreateInstance(RuntimeType type, Boolean
publicOnly, Boolean& canBeCached,
RuntimeMethodHandleInternal& ctor)
   at System.RuntimeType.CreateInstanceSlow(Boolean publicOnly, Boolean
skipCheckThis, Boolean fillCache,
StackCrawlMark& stackMark)
   at System.RuntimeType.CreateInstanceDefaultCtor(Boolean publicOnly,
Boolean skipCheckThis, Boolean fillCache,
StackCrawlMark& stackMark)
   at System.Activator.CreateInstance(Type type, Boolean nonPublic)
   at Unosquare.Swan.Abstractions.SingletonBase`1.<>c.<.cctor>b__7_0()
   at System.Lazy`1.ViaFactory(LazyThreadSafetyMode mode)
   at System.Lazy`1.ExecutionAndPublication(LazyHelper
executionAndPublication, Boolean useDefaultConstructor)
   at System.Lazy`1.CreateValue()
   at Unosquare.Swan.Abstractions.SingletonBase`1.get_Instance()
   at Unosquare.RaspberryIO.Pi..cctor()
   --- End of inner exception stack trace ---
   at Unosquare.RaspberryIO.Pi.get_Camera()

Met vriendelijke groet,

Roy Wagemakers

2018-04-16 15:22 GMT+02:00 Roy Wagemakers <roywagemakers@gmail.com>:

geoperez, 2018-04-16:
Are you trying to run the app in Windows or MacOs?

RunnicFusion, 2018-04-16:
Hi,

Windows Iot 9latest build), developing torugh Visual Studio 2017.
.net core 2.0

Met vriendelijke groet,

Roy Wagemakers

RunnicFusion, 2018-04-16:
Hi,

I installed the release 2.0.x for arm 32 windows on the pi from here:
https://github.com/dotnet/core-setup/tree/master

Met vriendelijke groet,

Roy Wagemakers

2018-04-16 16:46 GMT+02:00 Roy Wagemakers <roywagemakers@gmail.com>:

geoperez, 2018-04-16:
Please check issue #14

closed 2018-04-16 by geoperez

RunnicFusion, 2018-04-19:
Thanks for the reply. But i'm using .net core console application, which is running trough powershel on my rasberry pi 3b.
I can control the gpio the way it should but i also need access to my camera module (and later rfid reader). 

Your github repo say's:
" The Raspberry Pi's IO Functionality in an easy-to-use API for .NET (Mono/.NET Core). Our mission is to make .NET a first-class citizen in the Python-centric community of Raspberry Pi developers. "

What can I do to make this work?

geoperez, 2018-04-19:
How are you getting access to the GPIO? Using this powershell module https://github.com/PowerShell/PowerShell-IoT?

RunnicFusion, 2018-04-19:
No, doing it this way:

gpio = GpioController.GetDefault();

var pin = gpio.OpenPin(4, GpioSharingMode.Exclusive);
pin.SetDriveMode(GpioPinDriveMode.Output);

From the Windows.Devices.DevicesLowLevelContract assembly.