unosquare/raspberryio #49: Issue with TemperatureSensorAM2302 on May 27 commit

michaeldimoudis, 2018-05-10:
Hi there, firstly thanks for this great library.

I'm using the TemperatureSensorAM2302, even though it's not that accurate it used to work. The update on May 27 https://github.com/unosquare/raspberryio/commit/880ec2c138f159a2e2c9869306fe01ec2b011151 broke it. Reverting back to before that commit still works for me.

The error on a RaspberryPi running Raspbian .NET Core 2.0.7 is:

ERR >> [TemperatureSensorAM2302] BAD CHECKSUM: Expected C0; Was: D0
WRN >> [TemperatureSensorAM2302] 0 = L:   50 | H:   24
WRN >> [TemperatureSensorAM2302] 0 = L:   53 | H:   23
WRN >> [TemperatureSensorAM2302] 1 = L:   51 | H:   70
ERR >> [TemperatureSensorAM2302] BAD CHECKSUM: Expected C0; Was: D0

geoperez, 2018-05-10:
Cc @mariodivece

isirour, 2018-09-04:
Hi all,

I am experiencing same error with:
- Unosquare.RaspberryIO.Peripherals 0.2.1
- Unosquare.RaspberryIO 0.17.0
- .NET Core 2.1.401

Any insight as to how to over the issue? 

Code snippet:
`var DataPin = Pi.Gpio[P1.Gpio22];
 var Sensor = new TemperatureSensorAM2302(DataPin);
 Sensor.OnDataAvailable += Sensor_OnDataAvailable;
 Sensor.Start();`

Code snippet handler:
`private static void Sensor_OnDataAvailable(object sender, TemperatureSensorAM2302.AM2302DataReadEventArgs e)
        {
            Console.WriteLine("Handler called, reading coming...");
            Console.Write(e.TemperatureCelsius);
            Console.Write("\n");
        }`
                
If I can be of any assitance, please let me know.

Cheers,
Isidro

michaeldimoudis, 2018-09-04:
Yea pity this hasn't been fixed, as simply reverting the code change as mentioned above gets the temperature sensor working again. I'm not using the nuget package in my project due to this.

geoperez, 2018-09-04:
My bad, i didn't revert the code tomorrow.

geoperez, 2018-09-06:
I revert the code and I published a new nuget (0.2.2). Can you try it?

isirour, 2018-09-06:
> Enviado el: jueves, 6 de septiembre de 2018 19:50
>
>
>
> I reverted the code and I published a new nuget (0.2.2). Can you try it?

Hi! 

 

I have updated the NuGet (now 0.2.2), recompiled, deployed and executed, but I experience the same error, all readings end up in Checksum error, as example:

 

22:31:16.573 INF >> Start Pulse Index: 2

22:31:16.674 WRN >> [TemperatureSensorAM2302] 0 = L:   51 | H:   25

-12,8

22:31:16.674 WRN >> [TemperatureSensorAM2302] 0 = L:   61 | H:   12

22:31:16.674 WRN >> [TemperatureSensorAM2302] 0 = L:   52 | H:   23

22:31:16.674 WRN >> [TemperatureSensorAM2302] 0 = L:   51 | H:   25

22:31:16.674 WRN >> [TemperatureSensorAM2302] 0 = L:   50 | H:   24

22:31:16.674 WRN >> [TemperatureSensorAM2302] 0 = L:   50 | H:   25

22:31:16.674 WRN >> [TemperatureSensorAM2302] 1 = L:   52 | H:   69

22:31:16.674 WRN >> [TemperatureSensorAM2302] 0 = L:   65 | H:   25

22:31:16.674 WRN >> [TemperatureSensorAM2302] 0 = L:   50 | H:   24

22:31:16.674 WRN >> [TemperatureSensorAM2302] 0 = L:   52 | H:   23

22:31:16.674 WRN >> [TemperatureSensorAM2302] 0 = L:   52 | H:   25

22:31:16.675 WRN >> [TemperatureSensorAM2302] 0 = L:   50 | H:   24

22:31:16.675 WRN >> [TemperatureSensorAM2302] 0 = L:   52 | H:   23

22:31:16.675 WRN >> [TemperatureSensorAM2302] 0 = L:   50 | H:   25

22:31:16.675 WRN >> [TemperatureSensorAM2302] 1 = L:   50 | H:   70

22:31:16.675 WRN >> [TemperatureSensorAM2302] 0 = L:   63 | H:   25

22:31:16.675 WRN >> [TemperatureSensorAM2302] 0 = L:   50 | H:   25

22:31:16.675 WRN >> [TemperatureSensorAM2302] 0 = L:   50 | H:   25

22:31:16.675 WRN >> [TemperatureSensorAM2302] 1 = L:   50 | H:   71

22:31:16.675 WRN >> [TemperatureSensorAM2302] 1 = L:   52 | H:   70

22:31:16.675 WRN >> [TemperatureSensorAM2302] 0 = L:   52 | H:   24

22:31:16.675 WRN >> [TemperatureSensorAM2302] 1 = L:   50 | H:   72

22:31:16.675 WRN >> [TemperatureSensorAM2302] 0 = L:   50 | H:   23

22:31:16.678 WRN >> [TemperatureSensorAM2302] Checksum: 68-02-00-00; Data: C0-A8-80-80-58

22:31:16.678 ERR >> [TemperatureSensorAM2302] BAD CHECKSUM: Expected 68; Was: 58

 

I wonder if maybe @michaeldimoudis can check and comment…

 

My setup is:

*	.NET 2.1
*	Unosquare.Raspberry.IO 0.17.0
*	Unosquare.RaspberryIO.Peripherals 0.2.2

 

I have tested the sensor with a DLL created based on the DHT22 code available online for Arduino and works ok, so it is something in the Unosquare.RaspberryIO.Peripherals code.

 

Complete code, in case I am making some mistake of my own:

 

using System;

using Unosquare.RaspberryIO;

using Unosquare.RaspberryIO.Gpio;

using Unosquare.RaspberryIO.Peripherals;

 

namespace WeatherConsole

{

    public class Program

    {        

        static void Main(string[] args)

        {

            try

            {

                var dataPin = Pi.Gpio[P1.Gpio22];

                var sensor = new TemperatureSensorAM2302(dataPin);

                sensor.OnDataAvailable += Sensor_OnDataAvailable;

                sensor.Start();

                Console.ReadKey();

            }

            catch (Exception e)

            {

                Console.Error.WriteLine(e.Message + "- " + e.StackTrace);

            }

        }

 

        private static void Sensor_OnDataAvailable(object sender, TemperatureSensorAM2302.AM2302DataReadEventArgs e)

        {

            Console.WriteLine("...Handler called, reading coming...");

            Console.Write(e.TemperatureCelsius);

            Console.Write("\n");

        }

    }

}

 

 

Regards,

Isidro

geoperez, 2018-09-06:
@michaeldimoudis can you show me the correct source code?

michaeldimoudis, 2018-09-06:
Hi @geoperez I don't see the reverted code in git yet? The git commit hash I am using that's working for me is 6e2c8ab30a6c00d20c131d3d538fdbc618f5aa47

It looks like simply reverting only the TemperatureSensorAM2302 code to that commit won't fix the issue. I am trying to find the exact commit in between 16/3/2018 and 27/3/2018 that broke the temperature sensor. I'll let you know when I find it.

michaeldimoudis, 2018-09-07:
Hi @geoperez @isirour I just created a PR with the fix. https://github.com/unosquare/raspberryio/pull/66

A simple revert didn't work, had to also remove the 101ms Sleep in `Unosquare.RaspberryIO/Gpio/GpioPin.cs`. I am not sure why a sleep was put in there in this commit https://github.com/unosquare/raspberryio/commit/48abfd2ff9a6404a17c73a8aae41797406be0a5c, so I am unaware if this will affect any other peripherals, as I can only test the AM2302 Temperature Sensor. I am now using this PR in my home app and GPIO pins + Temperature Sensor are working as expected.

geoperez, 2018-09-07:
New nuget is up: https://www.nuget.org/packages/Unosquare.RaspberryIO.Peripherals/0.2.3

Please let me know if it's working now.

michaeldimoudis, 2018-09-07:
Woohoo it's working now :) Thanks for pushing this fix up to nuget.

closed 2018-09-07 by michaeldimoudis

isirour, 2018-09-10:
It works like a charm ;) Thanks guys!