unosquare/raspberryio #51: Help with pca9685

i9connect, 2018-06-13:
I am trying to make commands via i2c, but I can not identify the deviceid. I would like to control servos using pca9685.

geoperez, 2018-06-14:
Hi, I'm not familiar with that board, but maybe you can post your source code here and see if anyone else give you some advice.

BTW I received your email, please keep posting on this thread because it's easier to keep track.

i9connect, 2018-06-14:
I would like to know if you can help me by using a feature of your code.
It's about i2c.

I have a PCA9685 connected to my raspberry, but at startup, it asks DeviceId.

I searched a lot before, but found no alternative to get the value.

When you run the i2cdetect -y command 1, the following value is returned:
0 1 2 3 4 5 6 7 8 9 a b c e f
00 - - - - - - - - - - - - - - -
10: - - - - - - - - - - - - - - - - - -
20: - - - - - - - - - - - - - - - - - -
30: - - - - - - - - - - - - - - - - - -
40: 40 - - - - - - - - - - - - - - - -
50: - - - - - - - - - - - - - - - - - -
60: - - - - - - - - - - - - - - - - - -
70: 70 - - - - - - - -

How do I get the DeviceId?
Thank you

geoperez, 2018-06-14:
Cc @mariodivece

i9connect, 2018-06-18:
News?

k3z0, 2018-10-19:
Hi @mbazzo,

The **DeviceId** it's actually the address of your device in the I2C bus. Accordinlly to your post, you have 2 available devices at **0x40** and **0x70** addresses (both in hexadecimal).

Accordinly to the **PCA9685** documentation the default address is **0x40**, so you can follow the [i2c example](https://github.com/unosquare/raspberryio#i2c-to-connect-ics) at the README using **0x40** as DeviceId.

mbazzo, 2018-12-11:
Hello, I need help with this board.
I'm resuming using lib.
I performed the following test:
                 var device = Unosquare.RaspberryIO.Pi.I2C.AddDevice (0x40);                
                 for (var i = 1; i <4095; i ++)
                 {
                     var t = Convert.ToByte (i);
                     device.WriteAddressWord (1, t);
                     Thread.Sleep (100);
                 }

but the LEDs do not make the intensity (PWM) and yes they are blinking.
Does anyone have code that works, or can you help me?

geoperez, 2018-12-11:
@k3z0 can you take a look?

k3z0, 2019-01-28:
@mbazzo 

This is quite difficult because we don't have a PCA9685 card. Thus, we can't run any kind of test.

I think, you should take a look in the PCA9685 documentation to figure it out which commands does it support, or which commands you must provide to the card in order to get it run properly.

If we can get access to a PCA9685 card we would gladly conduct some tests to help you. 

Greetings.

mbazzo, 2019-01-29:
@k3z0 
Thanks, my friend, 
I ended up searching and doing the code I published in GitHub (https://github.com/mbazzo/PCA9685)

It would be interesting to put this code in your library.

Hugs

dobova86, 2019-03-14:
> @mbazzo
>
> This is quite difficult because we don't have a PCA9685 card. Thus, we can't run any kind of test.
>
> I think, you should take a look in the PCA9685 documentation to figure it out which [...]

Hi @mbazzo, the i2cdetect sense 0x70 slave address, but it may be it's not true, it depends on the real state of hardware Ax (a0,a1,a2,a3,a4,a5) pins! My Pi sense 0x70, but my real addr is 0x7F.
Note that I can't send n-bytes to LED regs, but you need to write 2by2 using WriteAddressWord() call.

closed 2019-05-20 by stale[bot]