unosquare/raspberryio #57: Does this library support asp.net core 2.1?

Zeni241, 2018-07-07:
As I am getting bellow error as I reach sample code line in visual studio 2017. 
 ` var blinkingPin = Pi.Gpio[0];`
{System.TypeInitializationException: The type initializer for 'Unosquare.RaspberryIO.Pi' threw an exception. ---> System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation. ---> System.PlatformNotSupportedException: This library does not support the platform
   at Unosquare.RaspberryIO.Gpio.GpioController.Initialize(ControllerMode mode)
   at Unosquare.RaspberryIO.Gpio.GpioController..ctor()
   --- End of inner exception stack trace ---

geoperez, 2018-07-07:
Hi, that's because you need to run your app as admin under Linux. What
environment are you using?

Zeni241, 2018-07-08:
Hi
Thanks for reply. I am debugging the app in VS2017 on Windows 10. I want to put my app (along with this library) on raspberry Pi finally. Do I have to publish it there for this library to work?

geoperez, 2018-07-08:
Yes, you need to publish and copy the folder to your raspberry pi. You
can't debug with Windows, we have no way to emulate hardware like GPIO
inside Windows.

Zeni241, 2018-07-09:
Thanks a lot and good luck with your great work.

closed 2018-07-11 by geoperez