unosquare/raspberryio #58: WiringPi wrappers to its own project and NuGet package

robertsundstrom, 2018-07-09:
I think that the WiringPi library wrappers should be placed in their own project and distributed in its own NuGet package on which RaspberryIO has a dependency.

geoperez, 2018-07-09:
Any specific reason for that?

robertsundstrom, 2018-07-09:
@geoperez, I later updated the issue by adding the reason:

This would be great for those who would just want to use the wrapper (pure WiringPi functions)

geoperez, 2018-07-09:
We agree with that approach, in fact we are looking for support other
library (Pigpio https://github.com/unosquare/pigpio-dotnet). And the
ultimate goal is expose a public api to allow use any wrapper-connector,
but we have no enough time to achieve it right now.

closed 2019-02-04 by stale[bot]