unosquare/raspberryio #59: root privileges not required

berknam, 2018-07-11:
**Is your feature request related to a problem? Please describe.**
Currently the GPIO requires root privileges, but since this is using wiringPi and it no longer requires root privileges this shouldn't require them either.

**Describe the solution you'd like**
The GPIO shouldn't require root privileges.

**Describe alternatives you've considered**
I've tested removing the "IsRunningAsRoot" check before the "WiringPiSetup" and it works without using sudo.

**Additional context**
I don't know if the change to wiringPi works across all bords and revisions. I've only tested on RPi3. But if we test for the existence of "/dev/gpiomem/" should be enough to know that it can work without root privileges, if it doesn't exist we can keep the "IsRunningAsRoot" check.

geoperez, 2018-07-11:
@Israel-94 can you remove the check and test the playground with GPIO?

Israel-94, 2018-07-12:
@geoperez sure

Israel-94, 2018-07-12:
@berknam There is a new release, you can check it out [here](https://www.nuget.org/packages/Unosquare.Raspberry.IO/)

closed 2018-07-15 by geoperez