unosquare/raspberryio #61: Cannot catch HardwareException upon I2C Access to not Connected Device

MoppleTheWhale, 2018-07-15:
Hi,

I am accessing a PCF8574 I/O expander via I2C on a Raspbeery Pi. No problem, as long as the I/O expander is connected to the I2C bus. However, when it is not connected, the program crashes, although I am trying to catch the relevant exceptions.

I am using Visual Studio 2017, .Net 4.5.2, Unosquare.Raspberry.IO 0.15.1 and Unosquare.Swan 0.32.2, mono 4.6.2 and Raspbian 4.14.50.

The relevant code, which reproduces the problem is:

            I2CDevice pcf8574_0 = null;

            try {
               pcf8574_0 = I2CBus.Instance.AddDevice(0x20);
            }
            catch {
               logger?.Warn("Cannot access I2C bus");
            };

            try {
               pcf8574_0?.Write(0);
            }
            catch (Exception e) {
               logger?.Error($"cannot write to I/O expander: {e}");
            }

Executing the code, without the PC8574 connected to the I2C bus, results in the following stacktrace:

  at <unknown> <0xffffffff>
  at (wrapper managed-to-native) object.__icall_wrapper_mono_marshal_free (intptr) <0x00037>
  at (wrapper managed-to-native) Unosquare.RaspberryIO.Native.Standard.StrError (int) <0x0005f>
  **at Unosquare.RaspberryIO.Native.HardwareException..ctor (int,string) <0x00073>**
  **at Unosquare.RaspberryIO.Native.HardwareException.Throw (string,string) <0x00053>**
  **at Unosquare.RaspberryIO.Gpio.I2CDevice.Write (byte) <0x0006f>**
  at Linq2Sql_Test.Program.Main () <0x00117>
  at (wrapper runtime-invoke) object.runtime_invoke_void (object,intptr,intptr,intptr) <0x000bf>

Native stacktrace:

Debug info from gdb:

[New LWP 18424]
[New LWP 18425]
[New LWP 18426]
[New LWP 18427]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
0x76ea54cc in __waitpid (pid=18428, stat_loc=0x7ea29f88, options=0) at ../sysdeps/unix/sysv/linux/waitpid.c:29
29      ../sysdeps/unix/sysv/linux/waitpid.c: Datei oder Verzeichnis nicht gefunden.
  Id   Target Id         Frame
* 1    Thread 0x76fbc570 (LWP 18423) "Main" 0x76ea54cc in __waitpid (pid=18428, stat_loc=0x7ea29f88, options=0) at ../sysdeps/unix/sysv/linux/waitpid.c:29
  2    Thread 0x763ff470 (LWP 18424) "SGen worker" 0x76ea094c in __pthread_cond_wait (cond=0x3258f0, mutex=0x3258bc) at pthread_cond_wait.c:186
  3    Thread 0x76a5f470 (LWP 18425) "Finalizer" 0x76ea3014 in futex_abstimed_wait_cancelable (private=0, abstime=0x0, expected=1, futex_word=0x31d8c0) at ../sysdeps/unix/sysv/linux/futex-internal.h:205
  4    Thread 0x73c17470 (LWP 18426) "Main" 0x76e160f0 in read () at ../sysdeps/unix/syscall-template.S:84
  5    Thread 0x739ff470 (LWP 18427) "Timer-Scheduler" 0x76ea094c in __pthread_cond_wait (cond=0x1fd3930, mutex=0x1fd3914) at pthread_cond_wait.c:186

Thread 5 (Thread 0x739ff470 (LWP 18427)):
#0  0x76ea094c in __pthread_cond_wait (cond=0x1fd3930, mutex=0x1fd3914) at pthread_cond_wait.c:186
#1  0x0021cdfc in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

Thread 4 (Thread 0x73c17470 (LWP 18426)):
#0  0x76e160f0 in read () at ../sysdeps/unix/syscall-template.S:84
#1  0x7497868c in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

Thread 3 (Thread 0x76a5f470 (LWP 18425)):
#0  0x76ea3014 in futex_abstimed_wait_cancelable (private=0, abstime=0x0, expected=1, futex_word=0x31d8c0) at ../sysdeps/unix/sysv/linux/futex-internal.h:205
#1  do_futex_wait (sem=sem@entry=0x31d8c0, abstime=0x0) at sem_waitcommon.c:115
#2  0x76ea3158 in __new_sem_wait_slow (sem=0x31d8c0, abstime=0x0) at sem_waitcommon.c:282
#3  0x001ac174 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

Thread 2 (Thread 0x763ff470 (LWP 18424)):
#0  0x76ea094c in __pthread_cond_wait (cond=0x3258f0, mutex=0x3258bc) at pthread_cond_wait.c:186
#1  0x002158e8 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

Thread 1 (Thread 0x76fbc570 (LWP 18423)):
#0  0x76ea54cc in __waitpid (pid=18428, stat_loc=0x7ea29f88, options=0) at ../sysdeps/unix/sysv/linux/waitpid.c:29
#1  0x000bbb44 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

=================================================================
Got a SIGSEGV while executing native code. This usually indicates
a fatal error in the mono runtime or one of the native libraries
used by your application.
=================================================================

First of all I wonder why pcf8574_0 is not null, although the device is not connected. Then the call to I2CDevice .Write() raises an exception and  it looks like the constructor of the HardwareException is causing the crash. Any idea why this happens?

kind regards
Wolfgang

geoperez, 2018-07-16:
The `HardwareException` is thrown when the native code (wiringPi) is not able to connect to a device. The exception in the HardwareException constructor in the other hand, it's not clear why is happening. I need to review it.

geoperez, 2018-07-16:
Related issue with Mono (https://github.com/mono/mono/issues/9532)

geoperez, 2018-07-17:
Can you upgrade to version 0.17.0 (https://www.nuget.org/packages/Unosquare.Raspberry.IO/0.17.0)?

MoppleTheWhale, 2018-07-18:
After upgrading to version 0.17.0, the exception is caught as expected:
Unosquare.RaspberryIO.Native.HardwareException on I2CDevice.Write: (121) - A hardware exception occurred. Error Code: 121

This solves my problem. Thank you so much for your support!

closed 2018-07-18 by MoppleTheWhale