unosquare/raspberryio #62: libwiringPi.so exception

ggriffon, 2018-07-15:
Hi,
I am having this issue while using the PI GPIO :
System.DllNotFoundException: Unable to load shared library 'libwiringPi.so.2.46' 

Full exception :

`connection id "0HLFAL62TEK7E", Request id "0HLFAL62TEK7E:00000002": An unhandled exception was thrown by the application.
System.TypeInitializationException: The type initializer for 'Unosquare.RaspberryIO.Pi' threw an exception. ---> System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation. ---> System.DllNotFoundException: Unable to load shared library 'libwiringPi.so.2.46' or one of its dependencies. In order to help diagnose loading problems, consider setting the LD_DEBUG environment variable: liblibwiringPi.so.2.46.so: cannot open shared object file: No such file or directory`

geoperez, 2018-07-16:
Usually, the file `libwiringPi.so.2.46` is extracted from RaspberyIO and stored in the same path as your binary. Maybe a permission issue? What OS are you using?

Syndlex, 2018-07-18:
I have the same problem. I could fix it with compiling the wiringPi.so from source and move it to the dir with the binary’s.
Now I get the Error from WiringPi on stdout: `wiringPiISR: Can't find gpio program`

I am using the dotnet Standard version of the Lib.
I am using the newest 0.17.0 version from nuget
To Compile I use the crosscompiler from dotnet core.
The So file is missing in the Output path.

geoperez, 2018-07-18:
`gpio.2.44` is another library used by RaspberryIO.

Syndlex, 2018-07-18:
Jup what I mean with this. 

Both Libs are missing in the Output when using the Standard version of this RaspberryIO.

ggriffon, 2018-07-19:
I did the same coying and renaming the file. Is it supposed to be include in the package?

ggriffon, 2018-07-19:
VS 2018, win10
.net core 2.1 on raspbian.

k3z0, 2018-07-19:
Hi there,

Effectively **libwiringPi.so.2.46** and **gpio.2.44** are part of the RaspberryIO package, but they are embedded resources in the DLL, that's why you can´t find the files when exploring the nuget package.

The files are created when the **Pi** class is statically initialized, so the files must be created unless there are permissions issues.

We just performed some tests running the "playground" in a raspberry and there were no issues, the files were created as expected.

### Steps to test the playground project

1. Clone the github repository.
2. Deploy the project to the Raspberry using [sshdeploy](https://github.com/unosquare/sshdeploy)

    **`dotnet-sshdeploy push -c release -f netcoreapp2.0`**

    **Note:** [sshdeploy](https://github.com/unosquare/sshdeploy) will build the project and will deploy it to the raspberry accordingly to the configuration. See the github repository for more info.
3. Install .Net Core in the Raspberry.
4. Run the playgound project:

    `dotnet Unosquare.RaspberryIO.Playground.dll `

    You should see something like this:
[image]

If you lists the files in the published folder, you should see the 2 files created:

[image]

### Installing .Net Core 2.1 on the Raspberry Pi

 1. **`sudo apt-get install curl libunwind8 gettext`**
    This will use the apt-get package manager to install prerequiste packages.
 2. **`curl -sSL -o dotnet.tar.gz https://download.microsoft.com/download/4/0/9/40920432-3302-47a8-b13c-bbc4848ad114/dotnet-sdk-2.1.302-linux-arm.tar.gz`**
   This will download the latest .NET Core Runtime for ARM32.
 3. **`sudo mkdir -p /opt/dotnet && sudo tar zxf dotnet.tar.gz -C /opt/dotnet`**
    This will create a destination folder and extract the downloaded package into it.
 4. **`sudo ln -s /opt/dotnet/dotnet /usr/local/bin`** 
    This will set up a symbolic link (shortcut) to the dotnet executable.
 5. **`dotnet --info`**
    To test the installation. You should see something like this:

    [image]

Syndlex, 2018-07-21:
Oh this could be my problem.
My Runtime is in a Docker Container.
I dont know if the Dotnet Runtime can create files.

closed 2018-08-02 by geoperez

Bubelgin, 2019-08-28:
Having the same problem but with later versions, core 2.2.6:
Unhandled Exception: System.DllNotFoundException: Unable to load shared library 'libwiringPi.so.2.50' or one of its dependencies. In order to help diagnose loading problems, consider setting the LD_DEBUG environment variable: liblibwiringPi.so.2.50.so: cannot open shared object file: No such file or directory
   at Unosquare.WiringPi.Native.WiringPi.WiringPiSetupGpio()
   at Unosquare.WiringPi.GpioController.Initialize(ControllerMode mode)
   at Unosquare.WiringPi.GpioController..ctor()
   at Unosquare.WiringPi.BootstrapWiringPi.Bootstrap()
   at Unosquare.RaspberryIO.Pi.Init[T]()
   at RaspCore.Program.Main(String[] args) in E:\Projects\RaspCore\RaspCore\Program.cs:line 12

Line 12 is just: Pi.Init<BootstrapWiringPi>();

Can see the liblibwiringPi.so.2.50.so file on the raspberry after I execute the program even with sudo, but still get the same problem.
Raspberry Pi 3 B+, all latest versions for nugets and core runtime installed as of today.
Unosquare.Raspberry.Abstractions 0.4.0
Unosquare.Raspberry.IO 0.25.0
Unosquare.WiringPi 0.5.0

Bubelgin, 2019-08-28:
> 
> 
> you can just publish your app as "Self contained" so all dependencies will be packed inside the published file.
> target runtime = linux-arm
> deployment mode = self contained
> 
> if you are using net core 3.0, you can add these option in the project file and your ouput/publish file will be a single file :
> 
> ```
> <PublishSingleFile>true</PublishSingleFile>
> <PublishTrimmed>true</PublishTrimmed>
> ```

Absolutely the same problem even with dotnet-runtime-3.0.0-preview8-28405-07-linux-arm.tar.gz.
Published as self contained package, still run into same error: "Unable to load shared library 'libwiringPi.so.2.50' or one of its dependencies".
Did a "sudo chmod -R 755" for the whole folder and run the app with sudo, doesn't help.

sergio73, 2019-09-15:
Hello, I'm having the same problem, and  I think I found the reason about why this is happening, I have installed Raspbian Stretch and this version is only compatible with libc 6 (v2.24)

If I execute this to check the dependencies of libwiringPi I get this:
```
ldd libwiringPi.so.2.50

./libwiringPi.so.2.50: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.28' not found (required by ./libwiringPi.so.2.50)
        linux-vdso.so.1 (0x7eaef000)
        /usr/lib/arm-linux-gnueabihf/libarmmem.so (0x76eb6000)
        libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0x76e37000)
        libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0x76e0e000)
        librt.so.1 => /lib/arm-linux-gnueabihf/librt.so.1 (0x76df7000)
        libcrypt.so.1 => /lib/arm-linux-gnueabihf/libcrypt.so.1 (0x76db8000)
        libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0x76c79000)
        /lib/ld-linux-armhf.so.3 (0x76eea000)
```
This could mean that libwirignPi.so.2.50 was compiled using the last version of Raspbian which currently is "buster" and it was compiled using libc 6 (v2.28). 

The only way to solve this is to install libwiringpi from apt or to compile the library and copy/paste the .so file.

logicethos, 2020-01-28:
Same issue here.
I tried copying the .so file from Raspbian (/usr/lib/libwiringPi.so.2.50)
but now I get this:
`WiringPi does only support a simple interrupt callback that has no parameters.`