unosquare/raspberryio #65: Peripheral Classes Documentation

TheSleepyPenguin, 2018-08-02:
Hi, I'd like to use the 'Button' peripheral class (I've installed the `Unosquare.RaspberryIO.Peripherals` package), but I'm not sure how to use it as I can't seem to find any documentation for it.
I'm new to C# to apologies if my question seems a bit simple...
Thanks for helping me out! :)

geoperez, 2018-08-02:
Yeah, we don't have documentation for the peripherals yet. But the [Button source code](https://github.com/unosquare/raspberryio/blob/master/src/Unosquare.RaspberryIO.Peripherals/Button.cs) is really easy to read.

The class will initialize a button connected to a `GpioPin` and you need to set listeners to the events `Pressed` and `Released`. The Playground project has a [sample method](https://github.com/unosquare/raspberryio/blob/master/src/Unosquare.RaspberryIO.Playground/Program.Button.cs#L13) using a button (BCM Pin 26) to open a relay (BCM Pin 17)

```csharp
var relayChannel1 = Pi.Gpio.GetGpioPinByBcmPinNumber(17);
relayChannel1.PinMode = GpioPinDriveMode.Output;

var buttonPin = Pi.Gpio.GetGpioPinByBcmPinNumber(26);
buttonPin.PinMode = GpioPinDriveMode.Input;

var button = new Button(buttonPin);
button.Pressed += (s, e) => { "Pressed".Info(); };

button.Released += (s, e) =>
{
  "Released".Info();
  var value = relayChannel1.Read();
  relayChannel1.Write(!value);
};

Console.ReadKey();
```

TheSleepyPenguin, 2018-08-03:
Thanks for the reply, I'll try it out!

geoperez, 2018-10-15:
Can I close this issue?

TheSleepyPenguin, 2018-10-16:
@geoperez if you're asking me, then yes.

geoperez, 2018-10-16:
Yes, closing.

closed 2018-10-16 by geoperez