unosquare/raspberryio #67: Friendly PWM Gpio

juandrn, 2018-09-11:
I wan't to use this library because is very simple for digital pins, but I don't know how to use it with PWM pins to do in an easy way

Something like https://raspi.tv/2013/how-to-use-soft-pwm-in-rpi-gpio-pt-2-led-dimming-and-motor-speed-control.

Good job
Best regards
Thanks

SuperJMN, 2018-09-18:
I'm very interested in this :) I have installed the package and have seen that pins have some properties related to PWM, but I've found no documentation about them. I would like to setup the classic "breathing LED" example.

Thanks in advance!

geoperez, 2018-09-19:
@mariodivece would you recommend use [pigpio-dotnet](https://github.com/unosquare/pigpio-dotnet) to work with PWM?

juandrn, 2018-10-01:
I have an example I can do with few lines:

```csharp
  //I needed to do a map pin  for Raspberry Pi 3
  readonly Dictionary<int, int> pinMapper = new Dictionary<int, int> {
            { 0, 30 }, { 1, 31 }, { 2, 8 }, { 3, 9 }, { 4, 7 }, { 5, 21 },
            { 6, 22 }, { 7, 11 }, { 8, 10 }, { 9, 13 }, { 10, 12 },
            { 11, 14 }, { 12, 26 }, { 13, 23 }, { 14, 15 }, { 15, 16 },
            { 16, 27 }, { 17, 0 }, { 18, 1 }, { 19, 24 }, { 20, 28 },
            { 21, 29 }, { 22, 3 }, { 23,4 } ,{ 24, 5 }, { 25, 6 },
            { 26, 25 }, { 27, 2 }
        };
  if (!pinMapper.TryGetValue(YOUR_PIN, out int value))
  {
        throw new InvokerException("Invalid Pin", InvokerExceptionType.Warn);
  }
  var pin = Pi.Gpio[value];
  pin.PinMode = Unosquare.RaspberryIO.Gpio.GpioPinDriveMode.PwmOutput;
  pin.PwmRegister = YOUR_VALUE;
  ...
```

It work only for pins 13 and 18

juandrn, 2018-10-02:
@geoperez Is there something like this example in pigio-dotnet?

Simplified code... find in: [rpi gpio code samples](https://elinux.org/RPi_GPIO_Code_Samples)

``` CPP

/*
   pulse.c
   gcc -o pulse pulse.c -lpigpio -lrt -lpthread
   sudo ./pulse
*/

#include <stdio.h>
#include <pigpio.h>

int main(int argc, char *argv[])
{
   double start;

   if (gpioInitialise() < 0)
   {
      fprintf(stderr, "pigpio initialisation failed\n");
      return 1;
   }

   /* Set GPIO modes */
   gpioSetMode(17, PI_OUTPUT);

   /* Start 75% dutycycle PWM on GPIO17 */
   gpioPWM(17, 192); /* 192/255 = 75% */

   /* Stop DMA, release resources */
   gpioTerminate();

   return 0;
}

```
I only see a blink gpio example in [pigpio-dotnet](https://github.com/unosquare/pigpio-dotnet)

k3z0, 2018-10-04:
Hi, 

here is a simple code for led dimming using RaspberryIO:

```csharp
   var pin = Pi.Gpio[P1.Gpio18];
   pin.PinMode = GpioPinDriveMode.PwmOutput;
   pin.PwmMode = PwmMode.Balanced;
   pin.PwmClockDivisor = 2; 
   while (true)
   {
      for (int x = 0; x <= 100; x++)
      {
         pin.PwmRegister = (int)pin.PwmRange / 100 * x;
         Thread.Sleep(10);
      }

      for (int x = 0; x <= 100; x++)
      {
         pin.PwmRegister = (int)pin.PwmRange - ((int)pin.PwmRange / 100 * x);
         Thread.Sleep(10);
      }
   }
```

[image]

**PwmRange** is the maximun value of the pulse width, than means 100% of pulse width. Changing this value allows you to have a more fine or coarse control of the pulse width (default 1024).

**PwmRegister** is the current pulse width. Changing this value allows you to change the current pulse width and thus the duty cycle.

**Duty Cycle** is equals to **PwmRegister** divide by **PwmRange**, assuming a **PwmRange** value of 1024 (default), we have:

| PwmRegister | Duty Cycle | 
| :---: | :---: |
| 0 | 0% |
| 256 | 25% |
| 512 | 50% |
| 768 | 75% |
| 1024 | 100% |

juandrn, 2018-10-04:
Thanks @k3z0 for the answer, the first example that I wrote works same as your example, but could you change gpio to another distinct of 18 and 21? because that is the only problem that I have

closed 2018-10-05 by juandrn