unosquare/raspberryio #74: Enabled getting a jpeg from the video stream

dommy1C, 2018-10-29:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

k3z0, 2018-11-05:
Hi @dommy1C

The Pi captures video as a raw H264 video stream, so in order to obtainan an image from video stream you must decode the raw H264 bytes in the **onDataCallback** callback. This is not an easy task, so maybe you can search for a third party library for H264 decoding.

As an alternative you can use **[Pi.Camera.CaptureImage](https://github.com/unosquare/raspberryio#capturing-images)** using a sufficient refreshing rate for simulate video.

Greetings.

k3z0, 2018-11-06:
Hi @dommy1C,

The default frame rate is 25 fps, you can change the frame rate setting a value between 2 and 30 to the **CaptureFramerate** property in the **CameraVideoSettings** object that you pass to **Pi.Camera.OpenVideoStream**.

dommy1C, 2018-11-06:
Hi

and how would i take a jpeg at that same rate?

thanks

k3z0, 2018-11-06:
As I told you, you must decode the raw H264 bytes in the **onDataCallback** callback, that way you can obtain all the frames.

Or you can run **Pi.Camera.CaptureImage** maybe using a loop with a delay, that way you can obtain a jpg every cicle (in this case, you must test the number of frames you can get every second, setting different delay times, to achieve a nice performance):

```csharp
   while (true)
   {
      await Task.Delay(100);
      var _jpgBytes = Pi.Camera.CaptureImage(_settings);
   }
```
With above code you can get up to 10 fps.

dommy1C, 2018-11-07:
Hi

Thanks for enduring my questions.

I know what you said about the onDataCallback and I know you said it will be a lot of work to get this working. So, firstly, I wanted to see if I could get the CaptureImage to work as it would obviously be simpler to code.

I know all about timers and the code you supplied etc (thanks).

But, the following code does NOT give 10fps.  Just because you coded it as such does not  mean it will give you the desired results.  It takes time for the software to take that picture. Like I said it gives me 1 frame per sec (at best).

So, I will just have to work with the onDataCallback  event.

Thanks for your time.

Appreciated.

Andrew

k3z0, 2018-11-07:
Hi Andrew,

Well, this is true, yo must run some tests with **CaptureImage**, try to play with the settings (image size, quality, etc), may be this way you can get more fps.

As you said, I think your best approach is to work with the **onDataCallback** trying to decode the raw H264. If you find a nice library or another solution, please share it with us.

closed 2019-01-28 by stale[bot]