unosquare/raspberryio #79: LCD 16x2

mbazzo, 2018-12-11:
Has anyone ever needed to write 16x2 using a library?
How to make?

geoperez, 2018-12-12:
Sample code for a Oled using pigpio-dotnet (similar project to this one):

https://github.com/unosquare/pigpio-dotnet/blob/master/Unosquare.PiGpio.Workbench/Runners/PiOled.cs

mbazzo, 2018-12-17:
I'm using this component

"https://www.banggood.com/en/LCD1602-Adapter-I2CIICTWI-Serial-Interface-Module-Board-For-Arduino-p-1056942.html?gmcCountry=BR&currency=BRL&createTmp=1&utm_source=googleshopping&utm_medium=cpc_elc&utm_content=frank&utm_campaign = pla-he-br-pc-0514 & gclid = EAIaIQobChMIifHo8sGn3wIVD4WRCh0Y9wpHEAQYAiABEgLL5fD_BwE & cur_warehouse = CN "

closed 2019-02-22 by stale[bot]