unosquare/raspberryio #82: pigpio wave_add_generic translation in unosquare.raspberryio?

Rainson12, 2019-01-13:
Im currently trying to translate a python script to .net core. However im failing because im unsure about what pigpios wave_add_generic does and what the translation in raspberryio would look like?

In python i have the following snippit which i want to translate (minified):
   ```
pi = pigpio.pi() 
   wf=[]
   wf.append(pigpio.pulse(1<<TXGPIO, 0, 9415))
   wf.append(pigpio.pulse(0, 1<<TXGPIO, 89565))
   pi.wave_add_generic(wf)
   wid = pi.wave_create()
   pi.wave_send_once(wid)
```

how would this look like in raspberryio?

geoperez, 2019-01-14:
RaspberryIO is still using wiringPi, but we have a wrapper around PiGpio at https://github.com/unosquare/pigpio-dotnet where the Waveform methods are implemented.

closed 2019-01-14 by geoperez

Rainson12, 2019-01-14:
@geoperez would you mind releasing/publishing the pigpio-dotnet library to nuget?