unosquare/raspberryio #92: GPIO Pins digital read and write doesn't work on Raspberry Pi 3 Model B+

oblomingo, 2019-02-05:
**Describe the bug**
I'm developing .net core application to control 5V relay with Raspberry Pi 3 Model B+ and I can't control 5v relay.  

**To Reproduce**
I have VMA406 (https://www.lemona.lt/LIUSE/NI/En/Pdf/vma406_a4v01.pdf) relay.
I connected 3 input contacts as described in the instruction:
1). S - to Gpio 26 (Physical pin 37)
2). + - to 5V Power (Physical pin 2)
3). - - to Ground (Physical pin 14)
I used this Raspberry Pi pinout scheme https://pinout.xyz/pinout/

Then I created simple console application with code:
```
Pi.Gpio.Pin26.PinMode = GpioPinDriveMode.Input;
// The below lines are reoughly equivalent
var isOn = Pi.Gpio.Pin26.Read(); // Reads as a boolean
var isOnText = isOn ? "on" : "off";
Console.WriteLine($"Gpio26 is {isOnText}");

Console.ReadLine();

Console.WriteLine($"Turnning on Gpio26");
Pi.Gpio.Pin26.PinMode = GpioPinDriveMode.Output;
// The below lines are reoughly equivalent
Pi.Gpio.Pin26.Write(true); // Writes a boolean
Console.WriteLine($"Turned on Gpio26");

Console.ReadLine();

Pi.Gpio.Pin26.PinMode = GpioPinDriveMode.Input;
// The below lines are reoughly equivalent
isOn = Pi.Gpio.Pin26.Read(); // Reads as a boolean
isOnText = isOn ? "on" : "off";
Console.WriteLine($"Gpio26 is {isOnText}");

Console.ReadLine();

Console.WriteLine($"Turnning off Gpio26");
Pi.Gpio.Pin26.PinMode = GpioPinDriveMode.Output;
// The below lines are reoughly equivalent
Pi.Gpio.Pin26.Write(GpioPinValue.Low); // Writes a boolean
Console.WriteLine($"Turned off Gpio26");

Console.ReadLine();

Pi.Gpio.Pin26.PinMode = GpioPinDriveMode.Input;
// The below lines are reoughly equivalent
isOn = Pi.Gpio.Pin26.Read(); // Reads as a boolean
isOnText = isOn ? "on" : "off";
Console.WriteLine($"Gpio26 is {isOnText}");

Console.ReadLine();
```

**Expected behavior**
After writing values to Pin26 output contacts should be opened/closed.
But output contacts doesn't change. Contacts C - CO - always opened. Contacts C - NC - always closed,  regardless of when pin26 value was written. By the way TemperatureSensorAM2302 also connected to the Raspberry and it's working perfectly with your library. 

**Devices: **
Raspberry Pi 3 Model B+
VMA406 relay.

k3z0, 2019-02-05:
Hi @oblomingo 

I think that, because the RaspberryIO version you are using, you should use the WiringPi pin numbering system. In WiringPi numbering system pin26 is [BMC12](https://pinout.xyz/pinout/pin32_gpio12#) (physical pin 32). Could you please try to use this pin or, if you need to use the physical pin 37, try to use the pin25 ([BCM26](https://pinout.xyz/pinout/pin37_gpio26#)).

This is because that version of RaspberryIO was totally based on WiringPi and, thus, it uses WiringPi pin numbering system.

Try to change the pin and let me know if it works.

oblomingo, 2019-02-06:
Hi @k3z0 !
Now it's working like a charm! Thanks a lot!
Could you please close this issue?

closed 2019-02-06 by oblomingo