unosquare/raspberryio #93: Problem with RaspberryIO v0.18.0

DanielSSilva, 2019-02-11:
I'm currently trying to migrate [this](https://github.com/techyian/RPi.SenseHat) project to fully support dotnet core. It currently has changes on both wiringPi and this project. I'm trying to remove those changes (because that wouldn't allow to keep versions up to date).
The current state can be seen [here on this "demo project"](https://github.com/DanielSSilva/MyCSharpSenseHat). Currently it compiles, but when I launch the Rpi.SenseHat.Tools it throws an exception:
```
Unhandled Exception: System.IO.FileLoadException: Could not load file or assembly 'Unosquare.Raspberry.Abstractions, Version=0.2.1.0, Culture=neutral, PublicKeyToken=null'. The located assembly's manifest definition does not match the assembly reference. (Exception from HRESULT: 0x80131040)
   at Emmellsoft.IoT.Rpi.SenseHat.SenseHatFactory.CreateDisplayJoystickI2CDevice()
   at Emmellsoft.IoT.Rpi.SenseHat.SenseHatFactory.GetSenseHatTask() in /home/pi/sensehat_csharp/Rpi.SenseHat/SenseHatFactory.cs:line 52
   at Emmellsoft.IoT.Rpi.SenseHat.Tools.Program.Main(String[] args) in /home/pi/sensehat_csharp/Rpi.SenseHat.Tools/Program.cs:line 44
   at Emmellsoft.IoT.Rpi.SenseHat.Tools.Program.<Main>(String[] args)
```
I've tried to implicitly include the nuget package `<PackageReference Include="Unosquare.Raspberry.Abstractions" Version="0.2.1" />` to target 0.2.1 on all projects, but it still complains about the same.
I navigated to the code (on Rpi.SenseHat > SenseHatFactory.cs > CreateDisplayJoystickI2CDevice) and when I navigate to the "I2C" on  `var device = Pi.I2C.AddDevice(0x10);` it shows me `#region Assembly Unosquare.Raspberry.Abstractions, Version=0.2.0.0, Culture=neutral, PublicKeyToken=null` which is the version 0.2.0.0 instead of 0.2.1.0.

I'm not really familiar with dll versioning. How is this possible? Also, RaspberryIO v0.18 requires Abstractions Version 0.2.1.0, so I'm really confused here.
If I'm not mistaken, you don't need to have the SenseHat to reproduce this.
I'm posting this here because to me, it seems that "something" inside your projects is still referring the abstractions 0.2.0.0. 

**To Reproduce**
Steps to reproduce the behavior:
1. Clone [this](https://github.com/DanielSSilva/MyCSharpSenseHat) project 
2. compile and run

**Devices**
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 9.4 (stretch)
Release:	9.4
Codename:	stretch

SenseHat

As for additional context, I've tried to compile and publish on my linux machine (dotnet publish -r linux-arm) and on my pi, both cases without any success

paez9817, 2019-02-11:
Hi, I have the same problem updating from 0.14.0 to 0.18.0 on raspberry

dobova86, 2019-02-11:
Same problem with new Console project :

> System.IO.FileLoadException: Could not load file or assembly 'Unosquare.Raspberry.Abstractions, Version=0.2.1.0, Culture=neutral, PublicKeyToken=null'. The located assembly's manifest definition does not match the assembly reference.

I checked Raspberry.IO, RaspberryIO.Peripherals and WiringPi but all reference 0.2.1 Raspberry.Abstraction.

DanielSSilva, 2019-02-11:
What's intriguing me here is the "The located assembly's manifest definition does not match the assembly reference". I don't understand much about dll and versioning, but by reading that error, seems that the assembly is being defined to refer to v0.2.0 somewhere. What I'll try to do later is to compile the raspberryio locally and point my projects to use the local version instead of the nuget package. 

As a side note, the namespace is Unosquare.RaspberryIO.Abstractions but the nuget package is Unosquare.Raspberry.Abstractions. Although it seems that the naming has nothing to do with the problem (looks like previous version were already like this) this naming convention is a little bit confusing

dobova86, 2019-02-11:
If this can help, the physical dll file is _Unosquare.Raspberry.Abstraction.dll_,  but the file version showed by file details is **2.0.0**

DanielSSilva, 2019-02-11:
Right, which means that the build process is generating the version 0.2.0 instead of the 0.2.1, although the package explicitly references the version 0.2.1

dobova86, 2019-02-11:
[image]

DanielSSilva, 2019-02-11:
But on the left it says 0.2.1. So **_maybe_** it means that the nuget version is 0.2.1 but the assembly is still 0.2.0. Can anyone please confirm this or correct me if I'm wrong?

lucam5, 2019-02-12:
The nuget package has the wrong 0.2.0 dll.
If you clone the master project and just compile the Abstraction project, you can use the dll generated inside bin folder. It compiles with the right version number. Here is attached the 0.2.1 dll waiting an update of the nuget package.
[Unosquare.Raspberry.Abstractions.dll.zip](https://github.com/unosquare/raspberryio/files/2856608/Unosquare.Raspberry.Abstractions.dll.zip)

dobova86, 2019-02-12:
> The nuget package has the wrong 0.2.0 dll.
> If you clone the master project and just compile the Abstraction project, you can use the dll generated inside bin folder. It compiles with the right [...]

Oh thanks a lot...

k3z0, 2019-02-13:
Hi @DanielSSilva,

We just release some updates to our nugets packages in order to solve the versioning problems.

- **Unosquare.Raspberry.IO** 0.18.1
- **Unosquare.RaspberryIO.Peripherals** 0.2.4
- **Unosquare.WiringPi** 0.2.4

I made some changes to your project ([here](https://github.com/k3z0/MyCSharpSenseHat)), but I'm not be able to test it because I don't have the hardware. Could you please test it and let me know if it worked?

Greetings.

paez9817, 2019-02-14:
Thanks @k3z0, It works on raspberry

geoperez, 2019-02-14:
Can I close this issue?

DanielSSilva, 2019-02-14:
hello @k3z0 . Thanks for the feedback. I'll test it

@geoperez I think you can. If something doesn't work I can re open it

closed 2019-02-14 by geoperez