unosquare/raspberryio #94: Breaking change in Unosquare.Raspberry.IO (0.18)

forki, 2019-02-11:
**Describe the bug**

After update to Unosquare.Raspberry.IO (0.18) I can't find the Pin values anymore. Docs show `WiringPiPin.Pin00`  as example, but can't find that one as well.

[image]

DanielSSilva, 2019-02-11:
Seems like this version "broke" quite a few things. The reference to the Abstractions namespace is broken as well #93

DanielSSilva, 2019-02-11:
I'm commenting here because on the bug that I've just mentioned, it also uses the Unosquare.RaspberryIO.Pi class

geoperez, 2019-02-14:
@forki please check issue #93

k3z0, 2019-02-14:
Hi @forki 

At the beginning, RaspberryIO was built around WiringPi library and all our classes, properties, enums, etc. was based into those ones used in WiringPi too. 

Now, we are working on a more general version of RaspberryIO (Abstractions) that, could use any core library (WiringPi, PiGpio or even new ones). So, it was necessary to change certain properties and enums for more general ones.

In order to use the new RaspberryIO version you need to follow the next steps:
- Add the reference to the [RaspberryIO](https://www.nuget.org/packages/Unosquare.Raspberry.IO) nuget package.
- Add a reference to a valid implementation of **RaspberryIO.Abstractions**. For now, we have fully implemented the [WiringPi](https://www.nuget.org/packages/Unosquare.wiringpi) library and we are working in the [PiGpio](https://github.com/unosquare/pigpio-dotnet/) implementation.
- Bootstrap RaspberryIO with a valid implementation.
   Before start using RaspberryIO, you must initialize **Pi** class (bootstrapping process) with the valid Abstractions implementation, in order to let **Pi** know what implementation is going to use:
   ```csharp 
      Pi.Init<BootstrapWiringPi>();
   ```

**NOTE:** A breaking change in this new general version is the pinout numbering system. As we already explain above, RaspberryIO was using the WiringPi pinout numbering system, but now it uses the [BCM pinout numbering](https://pinout.xyz/#) system. The pin numbers are totally different in both systems, so we recommend you to double check carefully the physical pins where you connect any device.

geoperez, 2019-02-14:
@k3z0 update the README file with your explanation.

k3z0, 2019-02-18:
I have already updated the README with the new features and the procedure for up and running RaspberryIO.

geoperez, 2019-02-18:
Can I close this issue?

TylerLeonhardt, 2019-02-20:
@k3z0 does the `RaspberryIO.Abstractions` sustain the old pinout behavior if we pick WiringPi? Or is that abstraction layer for something else? 

I'm trying to not have a breaking change.

k3z0, 2019-02-20:
Hi @TylerLeonhardt,

Sorry, but the answer is no, RaspberryIO.Abstractions does not support the WiringPi pinout numbering system anymore.

The main target of `RaspberryIO.Abstractions` it is to support any third party library. At this moment, we only have finished the implementation of WiringPi and we are working in the implementation of PiGpio. Because of the general purpose of the library, we decided to use the BCM pinout numbering system, which is, actually, the numbering system using by the Microcontroller itself.

DanielSSilva, 2019-02-20:
Hi @k3z0 ,
Probably it's my lack of knowledge, but, if there's a Bootstrap that uses the wiringPi, shouldn't the pinout follow the bootstrap? Won't it be confusing to use a bootstrap from a library that uses one pinout but then you find out that it actually uses another?

k3z0, 2019-02-20:
Hi,

Well, right now, we only support the BCM pinout numbering system for `RaspberryIO.Abstractions`. We thought that, because it is a general abstraction, using a general numbering system was a good approach, however, accordingly to your comments, we decide that it will be a good idea the possibility to add support for multiple numbering systems, independent of the implementation.

Unfortunately, this change will take some time since we are very busy these days. So, I going to open a new ticket to follow up this new request.

k3z0, 2019-02-20:
Check issue #99 to follow up this new request.

geoperez, 2019-09-01:
@k3z0 are you working on this issue?

closed 2020-03-31 by geoperez