unosquare/raspberryio #95: Better explanation to what's bootstrap

DanielSSilva, 2019-02-13:
I see that I'm required to initialize the Pi class with some bootstrap. What's this bootstrap? I noticed that there's the "WiringPi" bootstrap. Are there others? I'm asking this because I'm having issues with I2C but I don't know if it's due to this bootstrap or something else.
I'm trying to add the senseHat to the I2C devices but getting an exception saying "Unable to select I2C device: Device or resource busy". But The python examples work perfectly.
Can you please provide more info about bootstrap?
Thanks!

k3z0, 2019-02-14:
Hi @DanielSSilva 

As you noticed, we need to improve the documentation for the new version. By the moment, you can check this answer https://github.com/unosquare/raspberryio/issues/94#issuecomment-463725349 for more information about using and bootstrapping of the new version.

k3z0, 2019-02-18:
I have already updated the README with the new features and the procedure for up and running RaspberryIO.

k3z0, 2019-03-11:
can we close this issue?

DanielSSilva, 2019-03-12:
Yes, thanks

closed 2019-03-12 by geoperez