unosquare/raspberryio #104: How to save camera stream as a file?

codpro2005, 2019-03-26:
I'm trying out to save the video stream of the ```Pi.Camera``` into a file by adding each byte of the callback data into a byte List and saving it as a .h264 video. Ideally the video should be of .mp4 format and therefore I'm running a bash script to convert the .h264 file into a .mp4 file.

My problem is that when I open the .mp4 video, though the first few seconds are fine, it ends up speeding up afterwards (and cutting the last 2 or so seconds out). I believe that the problem appears within saving the byte array as a .h264 but just for reference I'll put in the entire code:

```
using System;
using System.IO;
using System.Diagnostics;
using System.Collections.Generic;
using Unosquare.RaspberryIO;
using Unosquare.RaspberryIO.Camera;

namespace RaspberryIO
{
    class Program
    {
        static void Main(string[] args)
        {
            string h264File = "/home/pi/Videos/recording";
            string mp4File = "/home/pi/Videos/recording";
            TestCaptureVideo(h264File);
            H264ToMp4(h264File, mp4File, true);
        }

        static void TestCaptureVideo(string outputPath)
        {
            // Setup our working variables
            var videoByteCount = 0;
            var videoEventCount = 0;
            var startTime = DateTime.UtcNow;

            // Configure video settings
            var videoSettings = new CameraVideoSettings()
            {
                CaptureTimeoutMilliseconds = 0,
                CaptureDisplayPreview = false,
                ImageFlipVertically = true,
                CaptureExposure = CameraExposureMode.Night,
                CaptureWidth = 1920,
                CaptureHeight = 1080
            };
            List<Byte> bytes = new List<Byte>();
            try
            {
                // Start the video recording
                Pi.Camera.OpenVideoStream(videoSettings,
                 onDataCallback: (data) => {
                     videoByteCount += data.Length;
                     videoEventCount++;
                     foreach (var d in data)
                     {
                         bytes.Add(d);
                     }
                 },
                 onExitCallback: null);

                // Wait for user interaction
                startTime = DateTime.UtcNow;
                Console.WriteLine("Press any key to stop reading the video stream . . .");
                Console.ReadKey(true);
            }
            catch (Exception ex)
            {
                Console.WriteLine($"{ex.GetType()}: {ex.Message}");
            }
            finally
            {
                // Always close the video stream to ensure raspivid quits
                Pi.Camera.CloseVideoStream();

                // Output the stats
                var megaBytesReceived = (videoByteCount / (1024f * 1024f)).ToString("0.000");
                var recordedSeconds = DateTime.UtcNow.Subtract(startTime).TotalSeconds.ToString("0.000");
                Console.WriteLine($"Capture Stopped. Received {megaBytesReceived} Mbytes in {videoEventCount} callbacks in {recordedSeconds} seconds");
                File.WriteAllBytes($"{outputPath}.h264", bytes.ToArray());
            }
        }

        static void H264ToMp4(string inputPath, string outputPath, bool removeH264)
        {
            $"MP4Box -add {inputPath}.h264 {outputPath}.mp4".Bash();
            if (removeH264)
            {
                $"rm {inputPath}.h264".Bash();
            }
        }
    }

    public static class ShellHelper
    {
        public static string Bash(this string cmd)
        {
            var escapedArgs = cmd.Replace("\"", "\\\"");

            var process = new Process()
            {
                StartInfo = new ProcessStartInfo
                {
                    FileName = "/bin/bash",
                    Arguments = $"-c \"{escapedArgs}\"",
                    RedirectStandardOutput = true,
                    UseShellExecute = false,
                    CreateNoWindow = true,
                }
            };
            process.Start();
            string result = process.StandardOutput.ReadToEnd();
            process.WaitForExit();
            return result;
        }
    }
}
```

Any help would be gladly appreciated.

codpro2005, 2019-03-27:
UPDATE:

I tried out the exact same code today again and now it's working perfectly fine. I at first thought that it might've been a buffer problem but I tried to replicate this problem by having a lot more processes running than when it didn't work, with no success. It might be of use to point out that while the .mp4 file generated by using this code did speed up, it did not speed up when using raspivid through the terminal directly.

I'm not exactly sure though if this problem had anything to do with the assembly itself or if it was something completely different.

Feel free to close the issue at any point though since I don't have this problem anymore.

k3z0, 2019-03-27:
Hi @codpro2005 

We glad to hear that you don't have problems anymore. Maybe your code will be useful for other people trying to do the same. 

Thanks for sharing.

closed 2019-03-27 by k3z0

geoperez, 2019-03-27:
You can add it in the wiki!