unosquare/raspberryio #105: OledDisplaySsd1306: System.TypeInitializationException

dj-curls, 2019-04-07:
**Describe the bug**
` 12:38:30.892 ERR >> [PiOled] System.TypeInitializationException: The type initializer for 'Unosquare.RaspberryIO.Peripherals.OledDisplaySsd1306' threw an exception.
   at Unosquare.RaspberryIO.Peripherals.OledDisplaySsd1306..ctor(II2CDevice device, DisplayModel model, VccSourceMode vccSource)
   at controller_concept_one.PiOled.Setup() in C:\Users\Oliver\Documents\raspberry-controller\Raspberry Controller\PiOled.cs:line 33
   at Unosquare.Swan.Abstractions.RunnerBase.<Start>b__17_0()`

**To Reproduce**
Steps to reproduce the behavior:
1. Create new .net core project 
2. Add Unosquare.Raspberry.IO nuget package 
3. Add PiOled.cs from [https://github.com/unosquare/pigpio-dotnet/blob/master/Unosquare.PiGpio.Workbench/Runners/PiOled.cs](https://github.com/unosquare/pigpio-dotnet/blob/master/Unosquare.PiGpio.Workbench/Runners/PiOled.cs)
3.A. Change namespaces to match project, and check for null _display in Cleanup()
4. Run project on Pi

**Expected behavior**
No errors.

**Devices**
Raspberry Pi 3+ Raspbian Lite 2018-11-13 (Stretch, 4.14)

I think the error is occouring when dotnet is trying to load the embedded PNG file in the OledDisplaySsd1306.cs class in the constructor here:

`var resourceNames = typeof(OledDisplaySSD1306).Assembly.GetManifestResourceNames();
            var fontResource = resourceNames.First(r => r.Contains("AsciiFontPng"));
            using (var stream = typeof(OledDisplaySSD1306).Assembly.GetManifestResourceStream(fontResource))`

I've checked with Telerik's JustDecompile, and I can see the AsciiFontPng in the resources of the Unosquare.RaspberryIO.Peripherals.dll file.

geoperez, 2019-04-07:
Try to add the file from: https://github.com/unosquare/raspberryio/tree/master/src/Unosquare.RaspberryIO.Peripherals/Resources

geoperez, 2019-04-10:
???

closed 2019-07-02 by stale[bot]