unosquare/raspberryio #107: System.MissingMethodException

conhoj, 2019-04-08:
**Describe the bug**
Missing Method Exception when attempting to run Playground

**To Reproduce**
Steps to reproduce the behavior:
 mono Unosquare.RaspberryIO.Playground.exe
or 
 dotnet Unosquare.RaspberryIO.Playground.dll

get the same error:-

 18:58:34.137 INF >> Starting program at 08/04/2019 18:58:33
 18:58:34.319 ERR >> [Main] Error
    $type           : System.MissingMethodException
    Message         :
        Method not found: int Unosquare.RaspberryIO.Abstractions.Native.Standard.StringToInteger(string,intptr,int)
    TargetSite      : Void Bootstrap()
    StackTrace      :
        at Unosquare.WiringPi.BootstrapWiringPi.Bootstrap () [0x00010] in <8f45db76a1d248788faaef47a92f23ac>:0
        at Unosquare.RaspberryIO.Pi.Init[T] () [0x00024] in <9c2d149f4e0240fbb3ab95b3420dc46f>:0
        at Unosquare.RaspberryIO.Playground.Program.Main () [0x00048] in <392f8207c86d4f8d893229bb64150b60>:0
    Source          : Unosquare.WiringPi
    HResult         : -2146233069
 18:58:34.902 INF >> Program finished.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Devices**
Raspberry version V3 B Raspbian

geoperez, 2019-04-10:
Can you clone your repo again?

conhoj, 2019-04-11:
Hi done that works a charm Thankyou

closed 2019-04-11 by conhoj