unosquare/raspberryio #109: Unable to open SPI device: No such file or directory

Sofoklis4, 2019-04-14:
When  I try to use SPI the following error emerge:
"Unable to open SPI device: No such file or directory"
Any idea on this?

geoperez, 2019-04-15:
Can you post a sample code?

k3z0, 2019-04-16:
Have you already activated SPI in the Raspberry? If not, follow [this](https://www.raspberrypi-spy.co.uk/2014/08/enabling-the-spi-interface-on-the-raspberry-pi/) tutorial.

Sofoklis4, 2019-04-17:
`Pi.Spi.Channel0Frequency = SpiChannel.MinFrequency;

                byte[] response = new byte[3];
                byte[] request = new byte[3] { 0x01, 0x80, 0 };
                response = Pi.Spi.Channel0.SendReceive(request);

                int result = 0;
                result = response[1] & 0x03;
                result <<= 8;
                result += response[2];

                double moisture_percentage = (100 - ((result / 1023.00) * 100));
                weatherData.SoilMoisture = moisture_percentage;`

The "Pi.Init<BootstrapWiringPi>();" initialize the Pi.Init<BootstrapWiringPi>(); in the constructor of the controller

The SPI in the Ruspberry is enebled.

Sofoklis4, 2019-05-22:
any help on this issue? I tried several things but in vain.

k3z0, 2019-05-22:
Hi @Sofoklis4 

Do you still have the same problem "Unable to open SPI device: No such file or directory"?

Could you please add a diagram of your wiring to the MCP3008?

k3z0, 2019-05-22:
I'm getting that error only when SPI is not enabled or when the device file does not exist. Could you please tell me the list of devices that the raspberry lists to you when you run the next command:

```console
ls /dev/spi*
```

Sofoklis4, 2019-05-23:
Yes I receive the same error.
Thw diagram is the following
VDD     3V
VREF    3V
AGND   GND
CLK	    SCLK
DOUT   MISO
DIN	    MOSI
CS/SHDN	   CE0
DGND	GND
CH0	  A0 of sensor bme280

When I run the command ls /dev/spi* I receive the following message:
ls cannot access /dev/spi* no such file or directory

k3z0, 2019-05-23:
Because of the answer you are receiving when you run the **ls** command, I assume that your SPI is disabled. You should receive something like this:

[image]

In this case, you can see that there is 2 device available for SPI. Maybe you need to double check your SPI settings.

Sofoklis4, 2019-05-24:
Finally it seems that I found the mistake.
In the past I had installed drivers for an LCD 3.5" display for my raspi but I didnt use it any more.
I had never removed the drivers of the dispay from raspi so all the spi.
So I commented some lines at the config.txt related to the display and everything is fine.

THANKS FOR YOUR TIME AND YOUR HELP.
I really appreciate it.

Sofoklis

closed 2019-05-24 by Sofoklis4