unosquare/raspberryio #111: complete example

daveyostcom, 2019-04-22:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

daveyostcom, 2019-04-22:
Getting started with dotnetcore on RPi would be so much easier if the solution would include a console app project that builds and is ready to copy to RPi, along with a paragraph or two specifically about the specifics of the specific example. ;-)  It can just blink a light. Anything that goes the whole way there.

Thanks

k3z0, 2019-04-22:
Hi @daveyostcom,

We appreciate that you take your time to make suggestions to the project's readme. If you already had modified the readme with some improvements, please submit your PR so we can review your suggestions to add it to the project.

Thanks.

closed 2019-06-28 by stale[bot]