unosquare/raspberryio #112: Can

daveyostcom, 2019-04-22:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

daveyostcom, 2019-04-22:
Hi.  I can't see anything that helps with input from the pins. I'm trying to get input from the capacitive touch inputs on the [Explorer HAT Pro](https://shop.pimoroni.com/products/explorer-hat).

Thanks

k3z0, 2019-04-23:
Hi @daveyostcom 

We don´t have an "Explorer Hat Pro" for performing tests, but in its [technical reference page](https://github.com/pimoroni/explorer-hat/blob/master/documentation/Technical-reference.md), they mention that you must use I2C to communicate with the hat. In RaspberryIO's readme, you can found examples of [I2C](https://github.com/unosquare/raspberryio#i2c-to-connect-ics) communication.

closed 2019-06-29 by stale[bot]