unosquare/raspberryio #113: Error while loading shared library

gingters, 2019-05-02:
**Describe the bug**
I wanted to test RaspberryIO, but my test program is unable to load the `libwiringPi.so.2.46`. 

I double-checked, and both files `libwiringPi.so.2.46` and `gpio.2.44` are extracted to the file system when I execute my program. They are also both marked as executables.

**To Reproduce**
Steps to reproduce the behavior:
1. On the RaspberryPi install the dotnet core SDK 2.2
2. Create a new project file with the following content  
```
<Project Sdk="Microsoft.NET.Sdk">

  <PropertyGroup>
    <OutputType>Exe</OutputType>
    <TargetFramework>netcoreapp2.2</TargetFramework>
  </PropertyGroup>

  <ItemGroup>
    <PackageReference Include="Unosquare.Raspberry.Abstractions" Version="0.3.0" />
    <PackageReference Include="Unosquare.Raspberry.IO" Version="0.19.0" />
    <PackageReference Include="Unosquare.WiringPi" Version="0.3.0" />
  </ItemGroup>

</Project>
```
3.  Create a new file Program.cs with the following content:
```
using System;
using Unosquare.RaspberryIO;
using Unosquare.RaspberryIO.Abstractions;
using Unosquare.WiringPi;

namespace pigpiotest
{
    class Program
    {
        static void Main(string[] args)
        {
            Console.WriteLine("Hello World!");

            Pi.Init<BootstrapWiringPi>();

            var led = Pi.Gpio[BcmPin.Gpio05];
            led.PinMode = GpioPinDriveMode.Output;

            led.Write(true);
        }
    }
}
```
4. Call `dotnet run`

**Expected behavior**
The program should run, and turn one GPIO pin to HIGH (I have a LED connected to this pin).

**Experienced behaviour**
I get the following exception:
```
System.DllNotFoundException: Unable to load shared library 'libwiringPi.so.2.46' or one of its dependencies. In order to help diagnose loading problems, consider setting the LD_DEBUG environment variable: liblibwiringPi.so.2.46.so: cannot open shared object file: No such file or directory
   at Unosquare.WiringPi.Native.WiringPi.WiringPiSetupGpio()
   at Unosquare.WiringPi.GpioController.Initialize(ControllerMode mode)
   at Unosquare.WiringPi.GpioController..ctor()
   at Unosquare.WiringPi.BootstrapWiringPi.Bootstrap()
   at Unosquare.RaspberryIO.Pi.Init[T]()
   at pigpiotest.Program.Main(String[] args) in /home/pi/dotgpio/Program.cs:line 14
```

**Logs**
I followed the instructions and set `LD_DEBUG` on when executing. I attached the full log (`LD_DEBUG=all`, this made a 13MB log file which I zipped) and a second file with `LD_DEBUG=libs` only (114k) to this issue. However, I do not really know my way round Linux internals, so I can't find any helpful information in that file.

[lddebuglog.zip](https://github.com/unosquare/raspberryio/files/3139888/lddebuglog.zip)
[lddebuglog_libsonly.txt]https://github.com/unosquare/raspberryio/files/3139898/lddebuglog_libsonly.txt)

**Devices**
Raspberry Pi 3 B+

**Additional context**
After I typed this ticket, I tried one more thing:
I explicitly called `export LD_LIBRARY_PATH=$PATH:/home/pi/dotgpio/bin/Debug/netcoreapp2.2` before I called `dotnet run`.
This helped. However, it is pretty counter-intuitive to have to update `LD_LIBRARY_PATH` to my applications folder before executing my program.

k3z0, 2019-05-06:
Hi @gingters ,

I don't know if this behavior is caused because of how the run command actually works. Try to use publish command instead and run your app using the generated dll:

```
dotnet pigpiotest.dll
```

gingters, 2019-05-06:
I created a self-contained published project and executed the binary directly and that worked. Thanks for the hint.
That somehow makes the process of testing a bit slow when I build on the pi, but it works. Thanks for your answer!

closed 2019-05-06 by gingters

k3z0, 2019-05-06:
We have a custom project to deploy projects directly to the raspberry using SSH, try it out, is very cool and easy to use:

https://github.com/unosquare/sshdeploy