unosquare/raspberryio #114: GPIO Pin Interrupts: which pin? RegisterInterruptCallback overload throws exception.

jifka, 2019-05-03:
I can register an interrupt using a parameter-less callback function. But if I use the provided overload that supposedly returns the pin number and current value, an exception is thrown indicating that WiringPi does not support it.

This is using the latest official release of raspberryio.

k3z0, 2019-05-06:
Hi @jifka 

In fact, WiringPi does only support parameterless callback, the only one that actually supports parameters in the callback function is PiGpio. So, that overload is only supported when you use the PiGpio implementation.

jifka, 2019-05-06:
Thanks, and apologies for asking a simple question - how would I use the PiGpio implementation? Is this not yet released?

k3z0, 2019-05-06:
We are still working in the PiGpio implementation. For the moment you only will be able to use the parameterless callback function provides by WiringPi.

geoperez, 2019-07-05:
Is this issue WIP?

closed 2019-09-10 by stale[bot]