unosquare/raspberryio #116: MFRC522 Example

dstania, 2019-05-07:
Hey Guys,

could you include an example how to use the RFID Reader MFRC522?

Best regards.

k3z0, 2019-05-07:
Hi @dstania,

You can use the [Unosquare.RaspberryIO.Peripherals](https://www.nuget.org/packages/Unosquare.RaspberryIO.Peripherals) nuget package, we already have an implementation for MFRC522.

```csharp
var _reader = new RFIDControllerMfrc522();
while (true)
{
	if (_reader.DetectCard() == RFIDControllerMfrc522.Status.AllOk)
	{
		var uidResponse = _reader.ReadCardUniqueId();
		if (uidResponse.Status == RFIDControllerMfrc522.Status.AllOk)
		{
			var cardUid = uidResponse.Data;
			_reader.SelectCardUniqueId(cardUid);

			try
			{
				if (_reader.AuthenticateCard1A(cardUid, 7) == RFIDControllerMfrc522.Status.AllOk)
				{
					// Read or write data to sector
				}
			}
			finally
			{
				_reader.ClearCardSelection();
			}
		}
	}
}
```

**Note:** The default constructor uses the BCM22 pin.

A more detailed example could be found in the playground project ([Playgroud MFRC522](https://github.com/unosquare/raspberryio/blob/master/src/Unosquare.RaspberryIO.Playground/Program.Rfid.cs)).

dstania, 2019-05-08:
Hey @k3z0 ,
thank you very much! I invested lot of time for looking for exact that code! Now it is working!

Best regards.

closed 2019-05-08 by k3z0

dg71, 2019-07-08:
Hi @k3z0 ,

Could you please provide a wiring pin-out from MFRC522 module to Raspberry Pi 3+ to fit the above code?

Thank you !!

best regards Dieter

DavidStania, 2019-07-10:
Hi @dg71,
you can use  the attached wiring.
Best regards.
[image]

DavidStania, 2019-07-10:
Hey @k3z0,
i need one more time support for reading NFC Tags.
With the example code from Playground package, i can detect perfectly the id of a rfid card or nfc tag but i can't get the correct data of the blogs. If i scan the NFC tag with my smartphone, i'll get the correct data inside. Do you have an idee how to read all data inside all blogs of an tag?

Thank you and best regards.

k3z0, 2019-07-10:
Hi @DavidStania 

The example code in the Playground project reads all the blocks of data from an RFID card, here is what you get from the example:

[image]

About NFC tags, we haven't run enough tests since there are different kinds of tags. As far as I had read the RC522 does not support all kind of NFC tags, but I'm not pretty sure. I think that for NFC you need another chip like the **[PN532](https://www.nxp.com/docs/en/nxp/data-sheets/PN532_C1.pdf)**.

Greetings.