unosquare/raspberryio #117: Add netstandard 2.0 target

benscobie, 2019-05-09:
I noticed as part of the 0.19.0 release the netstandard target was changed to 1.3. This brings in a bunch of unneeded dependencies for those of us who can utilise netstandard 2.0:
[image]

It looks like you can simply re-add the netstandard 2.0 on top of netstandard 1.3 target to solve this and not cause any issues:
https://twitter.com/davidfowl/status/1008453902058917888
https://docs.microsoft.com/en-us/dotnet/standard/net-standard#which-net-standard-version-to-target

k3z0, 2019-05-14:
Hi @benscobie 

Thanks for your suggestion, we are going to add both target frameworks in order to avoid all the dependencies for those who can utilize netstandard 2.0.

k3z0, 2019-05-14:
We are going to release new nuget packages with the new target framework configuration.

k3z0, 2019-05-14:
We already release the nuget version 0.19.1, try it out, please.

benscobie, 2019-05-14:
I still have a similar dependency graph sadly. I had a deeper look and it appears the System.Net.Security dependency from Unosquare.Swan is bringing in quite a lot. No harm in the netstandard 2.0 target now being there though.

k3z0, 2019-05-15:
Hi @benscobie 

We already released a new [Swan nuget](https://www.nuget.org/packages/Unosquare.Swan) (version 1.2.3) we try to remove the most of additional dependencies. Could you please give it a try.

Thanks.

k3z0, 2019-05-20:
Hi @benscobie,

Have you already tested the new nuget package? 

Can we close this issue?

benscobie, 2019-05-20:
Hi,

Yes sorry I did. The dependency graph was still the same as before. When I installed Unosquare.Raspberry.IO it was bringing in Unosquare.Swan 1.2.0 which has the old netstandard 2.0 target dependencies. I'm not sure why it was doing that though as I would have thought the ">= 1.2.0" target would pick the latest.

geoperez, 2019-05-20:
Yeah, the latest nuget is still using Swan 1.2.0. Let me update the nuget.

geoperez, 2019-05-22:
The new nuget was published.

closed 2019-06-10 by geoperez