unosquare/raspberryio #136: GpioPin.Read() and GpioPin.Write() does not work on RaspberryPi 4

Zubini-1, 2019-07-19:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Devices**
Raspberry version and hardware devices.

**Additional context**
Add any other context about the problem here.

Zubini-1, 2019-07-19:
Describtion:
Gpiopin.Read() is always false
Gpiopin.Wirte(true) does not work.
I build a setup with a Led and a 330 ohm resitance to test the .Write(true) 
For the Read() test i connected the 3.3V directly to Pin 10.
I use RaspberryIO v0.23, WiringPi v0.3.1
I tested the numbering of WiringPi and BCM for the GPIO.
`Pi.Init<BootstrapWiringPi>();`
            `var Pin10 = (GpioPin)Pi.Gpio[10];`
            `Pin10.PinMode = GpioPinDriveMode.Input;`
            `bool Ergebnis = Pin10.Read();`
            `MessageBox.Show(Ergebnis.ToString())`
I use mono to execute the.exe and droped the dll files of RaspberryIO and WiringPI from the Windows machine in the same folder as the .exe

I tested my Setup with a python code directly on the Raspberry 4 (4GB Ram) and it has worked.
I also tested wether any function of RaspberryIO works. The function `Pi.Info.RaspberryPiVersion.ToString();` works fine.

The program was written in visual studio 17 on Windows 10. RaspberryI0 and WiringPi were added with Nuget in visual Studio.

I do not get any error messages.

k3z0, 2019-07-23:
Hi @Zubini-1 

Could you please tell me which version of Raspbaian are you using?

Zubini-1, 2019-07-23:
Raspbian GNU/Linux 10 (buster)

k3z0, 2019-07-23:
@Zubini-1 we found that WiringPI needs an update in order to work in Raspbian Buster, we are going to release a new version of [wiringpi-dotnet](https://github.com/unosquare/wiringpi-dotnet/) with this fix ASAP.

k3z0, 2019-07-30:
Hi @Zubini-1,

We recently released a new [nuget](https://www.nuget.org/packages/Unosquare.wiringpi) of [wiringpi-dotnet](https://github.com/unosquare/wiringpi-dotnet/) (0.4.0). We are using a new version of WiringPi (2.50) but not the latest version (2.52) yet, because it is not clear when the sources will be available.

Could you please give it a try in your Raspberry Pi 4 and let us know if it works properly?

Maybe during the day, we are going to release a preview version using the latest version of WiringPi, but since we make some little changes to the library and in order to do that we need the sources, it will remain in a preview state until we do the changes.

Zubini-1, 2019-07-30:
Hi,

thank you very much. Now it works. I updated with nuget the wiringpi-dotnet and copied the dll Files again in the execution directory.

closed 2019-07-30 by Zubini-1

k3z0, 2019-07-30:
Maybe you can try using our tool [ShhDeploy](https://github.com/unosquare/sshdeploy) to publish the files directly to the Raspberry Pi.