unosquare/raspberryio #140: How can i implement Dht11 class in my project?

dinoToribio, 2019-07-28:


Serk352, 2019-07-29:
Hi @dinoToribio , the easiest way to do it is using the class DhtSensor and his static method Create, inside the playground project there is a function called TestTempSensor, here you can see a working implementation of a Dht11 sensor.

Example:
https://github.com/unosquare/raspberryio/blob/master/src/Unosquare.RaspberryIO.Playground/Peripherals/Peripherals.TempHum.cs

dinoToribio, 2019-07-30:
Thank you Serk352.

dinoToribio, 2019-07-30:
By the way @Serk352, can you tell me the more details step by step on how i can run the project in a raspberry pi. And i use windows C# form application to use the class. Is that allowed? Sorry,  i asked a lot because i am just beginner in this kind of field.

Thank you in advance @Serk352.

Serk352, 2019-07-30:
Hi @dinoToribio ,
To run the project on the RaspberryPi you need to deploy the files to a local folder, you can accomplish this using our tool [sshdeploy](https://github.com/unosquare/sshdeploy). There is an [example ](https://github.com/unosquare/raspberryio#run-the-app-on-the-raspberry) in the RaspberryIO documentation.
About using winforms, apparently Net Core 3 will have support for it, but only on Windows. [More Info](https://docs.microsoft.com/en-us/dotnet/core/whats-new/dotnet-core-3-0). Running on Linux (Raspbian) it is only possible to use console apps.

Serk352, 2019-07-30:
Can we close this ticket?

k3z0, 2019-07-30:
Hi @dinoToribio 

Maybe you could use Winforms on Linux if you run your app using Mono, more info [here](https://www.mono-project.com/docs/gui/winforms/). Unfortunately, this is out of the scope of RaspberryIO.

dinoToribio, 2019-07-31:
I understand now. Thank you very much everyone.

closed 2019-07-31 by dinoToribio