unosquare/raspberryio #144: Web streaming

ffcruz85, 2019-08-12:
First of all, thanks for this great project.

I am trying to build web UI for basic control of the raspicam: video and still capture, and the ability to change basic camera settings (brightness, etc). All that seems pretty straightforward with this library. 

For this to be useful from a web UI, the user needs to be able to see a live stream of the camera feed. I know this is out of scope of this project per se, but perhaps you might have some suggestions, examples, or just pointing the way for any method where the callback of Pi.Camera.OpenVideoStream can be used to expose a web stream to be viewed on a browser? Just asking in case I am missing something obvious or maybe someone has implemented this using RaspberryIO

Thanks!

k3z0, 2019-08-13:
Hi @ffcruz85 

By the moment we don't have support for direct streaming from our library, instead, you need to use some other libraries, like FFmpeg.

Here are some examples I could find:
https://medium.com/@jaredtoddpetersen/home-security-live-streaming-with-raspberry-pi-f9293efca7ba
https://www.engineer2you.com/2016/10/rasbperry-pi-ffmpeg-install-and-stream.html
https://www.digikey.com/en/maker/blogs/streaming-live-to-youtube-and-facebook-using-raspberry-pi-camera

ffcruz85, 2019-08-20:
I figured, but wanted to make sure I was not missing a particular form of usage here. Thanks for your time and the links - great project!

closed 2019-08-20 by ffcruz85

gorillazliuzhe, 2019-10-28:
Hello. Excuse me, do you have an example to display video stream in web ui. Thank you.

gorillazliuzhe, 2019-10-28:
> I figured, but wanted to make sure I was not missing a particular form of usage here. Thanks for your time and the links - great project!

Hello. Excuse me, do you have an example to display video stream in web ui. Thank you.