unosquare/raspberryio #149: LED Strip Misssing

vbisbest, 2019-08-21:
I cannot seem to find the LED Strip (APA102) class under peripherals.  I am using the nuget package directly from nuget (not building myself).   Thanks!

geoperez, 2019-08-21:
What target framework are you using?

vbisbest, 2019-08-21:
.Net Core 2.2.  And targeting linux-arm (for Raspberry Pi 3B+)

geoperez, 2019-08-21:
Please upgrade to nuget version 0.5.0.

vbisbest, 2019-08-21:
Ok, the class is there now, but I cannot init PI.  Per your docs:

Pi.Init<BootstrapWiringPi>();

BootstrapWIringPi is not a valid object.

k3z0, 2019-08-21:
Hi @vbisbest 

Try to use [this ](https://github.com/unosquare/raspberryio#usage)init code:

```csharp
Pi.Init<BootstrapWiringPi>();
```

vbisbest, 2019-08-21:
Ok, I was missing that nuget package.  I have all that setup and now getting this error on the Pi.Init call.

fail: Microsoft.AspNetCore.Diagnostics.ExceptionHandlerMiddleware[1]
      An unhandled exception has occurred while executing the request.
System.IO.FileNotFoundException: Could not load file or assembly 'Unosquare.Swan, Version=1.2.4.0, Culture=neutral, PublicKeyToken=null'. The system cannot find the file specified.

File name: 'Unosquare.Swan, Version=1.2.4.0, Culture=neutral, PublicKeyToken=null'
   at Unosquare.WiringPi.BootstrapWiringPi.Bootstrap()
   at Unosquare.RaspberryIO.Pi.Init[T]()
   at LightWall.Controllers.HomeController.Index() in C:\Users\Ray\Documents\Projects\LightWall\Controllers\HomeController.cs:line 18
   at lambda_method(Closure , Object , Object[] )

vbisbest, 2019-08-21:
I should add, the version of swan that gets installed is 2.0.1 (seems to be looking for 1.2.4)

geoperez, 2019-08-21:
Thanks for the heads up. WiringPi nuget was outdated. Please upgrade to version 0.5.0.

vbisbest, 2019-08-21:
I think we are closer.  Now I get this error when I create the LED Strip.
Unable to open SPI device: No such file or directory

Here is the code I am using.  I tried SPI channel 0 and 1.
            
```
Pi.Init<BootstrapWiringPi>();
            Unosquare.RaspberryIO.Peripherals.LedStripAPA102C led = new Unosquare.RaspberryIO.Peripherals.LedStripAPA102C(10, 0, 1000000);
            led.ClearPixels();
```
And thank you very much for your prompt responses and fixes!

vbisbest, 2019-08-21:
AHHH! Never mind.  I had to enable the SPI interface on the rpi.   Thank you again!

closed 2019-08-21 by vbisbest