unosquare/raspberryio #173: Ssd1306 display error connected to i2c

yangsungyeol, 2019-10-03:
**To Reproduce**
Steps to reproduce the behavior:
1. I want to use the PiOled class included in Unosquare.RaspberryIO.Playground.Extra

2.   I instantiate it
If I try to call the setup onsetup () method
Error message is displayed

3. See error

Unhandled exception. System.Collections.Generic.KeyNotFoundException: The given key '60' was not present in the dictionary.
   at System.Collections.Generic.Dictionary`2.get_Item(TKey key)
   at Unosquare.WiringPi.I2CBus.GetDeviceById(Int32 deviceId)
   at linux_ds1631.Program.Main(String[] args) in E:\TEST_2015\linux_ds1631\Program.vb:line 79
Aborted

I am testing, just to drive the sample to the screen.

If I do not call the onsetup method, no error will occur but there will be no indication.
 If you provide a PiOled sample written in C # for me to test, I will try it.

**Screenshots**
I attached my system information and error screen.

 

**Devices**
I am a Raspberry Pi Model 3B
The OS is Ubuntu Mate 32 bit.

install .net core 3.0
I'm testing with VB.net.

The C # sample runs fine, but I can't find the piOled driven sample.

[image]
[image]

[image]

**Additional context**
Add any other context about the problem here.

geoperez, 2019-10-03:
@k3z0 can you take a look?

yangsungyeol, 2019-10-05:
Tell me what you need
I will do it

k3z0, 2019-10-07:
Hi @yangsungyeol 

Try changing the code in `OnSetup` function to use another overload of the `OledDisplaySsd1306 `constructor, the one that receives an instance of `II2CDevice`:

```csharp
protected override void OnSetup()
{
    var device = Pi.I2C.AddDevice(0x3c);
    _display = new OledDisplaySsd1306(
                           device,
                           OledDisplaySsd1306.DisplayModel.Display128X32,
                           OledDisplaySsd1306.VccSourceMode.Switching);
}
```

yangsungyeol, 2019-10-08:
It did not cause an error
I've modified the code as your instructions

It seems to work well, with no errors.
But still no message appears on my display.

I think the code is executed

Please tell me how to test more

[image]

yangsungyeol, 2019-10-08:
I will order another module and test it.
If you have a source that helps me, please provide

k3z0, 2019-10-08:
Hi @yangsungyeol 

For the moment we don't have an SSD1306 display to test the code. So it will be difficult for me to provide you some guidance. Nevertheless, that code was already tested and should work. 

As soon as I can get a display I will run some tests in order to help you.

closed 2020-04-23 by stale[bot]