unosquare/raspberryio #179: How to use VS2019 to remotely debug the console on the Raspberry Pi

gorillazliuzhe, 2019-10-18:
i need help .please

pgrawehr, 2019-10-19:
That's quite easy, actually, if you have the latest version of VS2019 installed. 
Make sure SSH is enabled on the Pi (use `sudo raspi-config` and `interfacing options`). Now just start your process (i.e. `dotnet myApp.dll`). 

Then, in Visual Studio, use Debug->Attach to process. Select SSH as transport, and enter "username@IP" in the address box (i.e. "pi@192.168.1.43"). Now a dialog opens where you can specify some connection details. When everything works, you will be presented with the process list on your raspberry. (Note: This works only if your process does not run as root, otherwise you have to use "root@IP" and previously set a SSH password for root).

gorillazliuzhe, 2019-10-20:
> That's quite easy, actually, if you have the latest version of VS2019 installed.
> Make sure SSH is enabled on the Pi (use `sudo raspi-config` and `interfacing options`). Now just start your process [...]

Thank you very much, but after executing dotnet myApp.dll, I can see this process in VS. But this console has already executed the last line of code. How can I restart the first line of code to start debugging. You can understand what I said
What does it mean?

pgrawehr, 2019-10-20:
There should be a way of starting the debugger together with your process, but I haven't found out how, yet. The easiest workaround is to (temporarily) introduce a `Console.ReadLine()` command as first step in your application, so that after starting it, it waits until you attach the debugger and then hit enter.

gorillazliuzhe, 2019-10-20:
> There should be a way of starting the debugger together with your process, but I haven't found out how, yet. The easiest workaround is to (temporarily) introduce a `Console.ReadLine()` command as [...]

ths,I tried to use Console.ReadLine() in the first line of the code; but still can't enter the program breakpoint, it's too bad.

pgrawehr, 2019-10-20:
When Visual Studio enters debugging mode, do your breakpoints (set one right after the Console.ReadLine() line) get enabled? Or do you see an exclamation point on the red circle to the left of the line? If so, there's something wrong with your setup, i.e. you are not copying the PDB files to your Pi.

gorillazliuzhe, 2019-10-23:
> When Visual Studio enters debugging mode, do your breakpoints (set one right after the Console.ReadLine() line) get enabled? Or do you see an exclamation point on the red circle to the left of the [...]

Thank you very much, I forgot the pdb file before. I can work now. Thank you again.

closed 2019-10-23 by gorillazliuzhe