unosquare/raspberryio #180: Using MCP23017 with I2C

dan1-de, 2019-10-20:
Hi, 

i want to use multiple I2C Port Expanders of MCP23017.
How can i access the GPIO Ports of the MCP23017 over I2C ?

Can you please provide an example ?

Thanks
Best regards Daniel

pgrawehr, 2019-10-20:
Have you read the data sheet? It's available here, for instance: https://produktinfo.conrad.com/datenblaetter/650000-674999/651440-da-01-en-IC_I_O_EXP__MCP23017_E_SP_SPDIP_28_MCP.pdf

dan1-de, 2019-10-21:
Ýes and after some tries i get the output working with this code:

### UnoSquareI2CDevice myDevice = Pi.I2C.AddDevice(0x20);
### myDevice.WriteAddressByte(0x14, 0x01);

0x20 is the Adress of the MCP23017 and 0x14 the GPIOA output latch.
With 0x01 i set the first pin GPA0 to high.

The question is now: Can i register any interrupts/callbacks over I2C if one of the GPIO Pins change or should i detect this with polling the registers all the time ?

k3z0, 2019-10-21:
Hi @dan1-de 

In the case you are exposing, I think your only choice is to polling your device's addresses for those inputs buttons.

pgrawehr, 2019-10-21:
The MCP23017 has two interrupt pins. Connect these to separate GPIO pins on the Pi and you should be able to use these as interrupt inputs (at least you get an interrupt if any of the pins changes).

dan1-de, 2019-10-22:
Thank you for your answers. 
I will use the interrupt pins. 

Best regards

Daniel

closed 2019-10-22 by dan1-de