unosquare/raspberryio #181: how to initialize a valid IBootstrap implementation.

THEsnarkysparky, 2019-10-22:
I have this in my main form

Pi.Init<BootstrapWiringPi>();

Using Raspian Buster.  Dotnet 4.61  

But the result is.........

Unhandled Exception:
System.InvalidOperationException: You must first initialize Pi referencing a valid IBootstrap implementation.
  at Unosquare.RaspberryIO.Pi.ResolveDependency[T] () [0x00007] in <0ea343189e2a42e7a36ec6a98a8451de>:0 
  at Unosquare.RaspberryIO.Pi.get_Gpio () [0x00000] in <0ea343189e2a42e7a36ec6a98a8451de>:0 
  at printercontrol.MainForm..ctor () [0x00016] in <3bcad23eb9424005a3f460b9a20bff98>:0 
  at (wrapper remoting-invoke-with-check) printercontrol.MainForm..ctor()
  at printercontrol.Program.Main (System.String[] args) [0x0000e] in <3bcad23eb9424005a3f460b9a20bff98>:0 
[ERROR] FATAL UNHANDLED EXCEPTION: System.InvalidOperationException: You must first initialize Pi referencing a valid IBootstrap implementation.
  at Unosquare.RaspberryIO.Pi.ResolveDependency[T] () [0x00007] in <0ea343189e2a42e7a36ec6a98a8451de>:0 
  at Unosquare.RaspberryIO.Pi.get_Gpio () [0x00000] in <0ea343189e2a42e7a36ec6a98a8451de>:0 
  at printercontrol.MainForm..ctor () [0x00016] in <3bcad23eb9424005a3f460b9a20bff98>:0 
  at (wrapper remoting-invoke-with-check) printercontrol.MainForm..ctor()
  at printercontrol.Program.Main (System.String[] args) [0x0000e] in <3bcad23eb9424005a3f460b9a20bff98>:0

SynergYFTW, 2019-10-22:
Pi.Init<BootstrapWiringPi>();

and add these using as well:
using Unosquare.RaspberryIO.Pi

SynergYFTW, 2019-10-22:
`Pi.Init<BootstrapWiringPi>();`

AlexanderJohnston, 2019-11-14:
I must be missing something. Everything else works fine but I can't call `Pi.Init<BootstrapWiringPi>();` without getting the following error:

```
The type or namespace name 'BootstrapWiringPi' could not be found (are you missing a using directive or an assembly reference?)
```

Any idea what I'm doing wrong? I'm sure there must be a using statement needed for it but Visual Studio isn't suggesting one.

AlexanderJohnston, 2019-11-14:
As usual, I spend 24 hours trying to resolve it on my own only to find the answer about 30 minutes after posting to the Issue.

I needed to add the WiringPi package.

AlexanderJohnston, 2019-11-14:
So it looks like I needed a reference to the WiringPi package. Seems to be working for now.

Zubini-1, 2019-12-11:
I have a simalar issue: 
            Pi.Init<BootstrapWiringPi>();
            var GPIO = Pi.Gpio[27];
It crashs at the second line with following error report:
System.InvalidOperationException: You need to load a valid assembly (WiringPi or PiGPIO).
  at Unosquare.RaspberryIO.Pi.ResolveDependency[T] () [0x00020] in <888f93d5325844e194afbf665f27ebc9>:0 

This code has already worked, but now it does not.

Have anyone an idea how to solve the problem. I have already updated in NuGet all unosquare add ins.

k3z0, 2019-12-12:
hi @Zubini-1 

Could you please share your project or tell us more information about the versions of the packages you are using?

Zubini-1, 2019-12-13:
Hi,

It runs with Mono on a Raspberry Pi 4b. On the Raspberry Pi is the latest WiringPi Version (2.52) installed.
In my C# Project i use UnoSquare.Raspberry.Abstractions v0.4.0, UnoSquare.Raspberry.IO v0.27.0, Unosquare.Swan v.2.4.3 and Unosquare.Swan.Lite v2.4.3, Unosquare.WiringPi v0.5.0. All installed with NuGet.

Here a bigger snippet of the code:
```
using Unosquare.RaspberryIO;
using Unosquare.RaspberryIO.Abstractions;
using Unosquare.WiringPi;

        public virtual void OpenComplete()
        {
            Pi.Init<BootstrapWiringPi>();
            var GPIO = Pi.Gpio[gpio_pin];
            GPIO.PinMode = GpioPinDriveMode.Output;
            GPIO.Write(false);
            status = false;
        }
```
Here the specs of the raspberry Pi:
```
pi@raspberrypi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

pi@raspberrypi:~ $ gpio -v
gpio version: 2.50
Copyright (c) 2012-2018 Gordon Henderson
This is free software with ABSOLUTELY NO WARRANTY.
For details type: gpio -warranty

Raspberry Pi Details:
  Type: Unknown17, Revision: 01, Memory: 0MB, Maker: Sony 
  * Device tree is enabled.
  *--> Raspberry Pi 4 Model B Rev 1.1
  * This Raspberry Pi supports user-level GPIO access.
```

k3z0, 2019-12-16:
Hi @Zubini-1 

Try to delete the references to the next nuget packages, since **UnoSquare.Raspberry.IO** and **Unosquare.WiringPi** already uses and download the versions they support.

- Unosquare.Swan
- Unosquare.Swan.Lite
- UnoSquare.Raspberry.Abstractions

Zubini-1, 2019-12-18:
Hi,

thank you for your help. The problem was solved with a downgrade of following NuGet Packages:

- Unosquare.Raspberry.Abstractions v0.4.0 -> v0.3.1
- Unosquare.Raspberry.IO v0.27.0 -> v0.24.0
- Unosquare.Swan v2.4.3 -> v2.3.1
- Unosquare.SwanLite v2.4.3 -> v2.3.1
- Unosquare.WiringPi v0.5.0 -> v0.4.0

k3z0, 2019-12-18:
Hi,

I think you need to remove the nuget packages I mention earlier, in that way you could use the last versions of **Unosquare.WiringPi** and **Unosquare.Raspberry.IO**.

**Unosquare.Raspberry.Abstractions** it's only necessary if you are going to implement a new library, actually **Unosquare.WiringPi** implements it.

I think the problem is caused because **Unosquare.Raspberry.IO** it's not compatible, yet, with the last versions of **Swan**. We are going to work on this issue.

closed 2020-02-29 by stale[bot]