unosquare/raspberryio #189: There is no StartSoftPwm method in IGpioPin?

feeho, 2019-11-20:
var range = 100;
var pin = Pi.Gpio[BcmPin.Gpio24];            
 pin.PinMode = GpioPinDriveMode.Output;
pin.StartSoftPwm(0, range);

I can not run the code and error message is: 'igpionpin' does not contain a definition of 'startsoftpwm'

k3z0, 2019-11-20:
Hi @feeho 

I think our documentation is outdated, we'll be working to fix it, sorry about it.

In order to the sample code to work you need to add the [wiringpi-dotnet](https://github.com/unosquare/wiringpi-dotnet/) [nuget](https://www.nuget.org/packages/Unosquare.wiringpi) package, which is an implementation of RaspberryPi abstractions.

You need to init RaspberryPi with wiringpi-dotnet and cast the _IGpioPin_ to the WiringPi _GpioPin_ implementation.

```csharp

using Unosquare.WiringPi;
...

Pi.Init<BootstrapWiringPi>();

var range = 100;
var pin = (GpioPin)Pi.Gpio[BcmPin.Gpio24];
pin.PinMode = GpioPinDriveMode.Output;
pin.StartSoftPwm(0, range);
```

In our playground project, you can find a [sample](https://github.com/unosquare/raspberryio/blob/ca08231ea71c88d75f8716ab4fa90ed001f089ba/src/Unosquare.RaspberryIO.Playground/Extra/Extra.Led.cs#L112) of using SoftPwm for dimming an LED.

geoperez, 2019-11-20:
@k3z0 can you update the documentation today, please?

feeho, 2019-11-21:
> Hi @feeho
> 
> I think our documentation is outdated, we'll be working to fix it, sorry about it.
> 
> In order to the sample code to work you need to add the [wiringpi-dotnet](https://github.com/unosquare/wiringpi-dotnet/) [nuget](https://www.nuget.org/packages/Unosquare.wiringpi) package, which is an implementation of RaspberryPi abstractions.
> 
> You need to init RaspberryPi with wiringpi-dotnet and cast the _IGpioPin_ to the WiringPi _GpioPin_ implementation.
> 
> ```cs
> using Unosquare.WiringPi;
> ...
> 
> Pi.Init<BootstrapWiringPi>();
> 
> var range = 100;
> var pin = (GpioPin)Pi.Gpio[BcmPin.Gpio24];
> pin.PinMode = GpioPinDriveMode.Output;
> pin.StartSoftPwm(0, range);
> ```
> 
> In our playground project, you can find a [sample](https://github.com/unosquare/raspberryio/blob/ca08231ea71c88d75f8716ab4fa90ed001f089ba/src/Unosquare.RaspberryIO.Playground/Extra/Extra.Led.cs#L112) of using SoftPwm for dimming an LED.

Thanks a lot

closed 2019-11-21 by feeho