unosquare/raspberryio #193: dotnet core asp webapi throws System.UnauthorizedAccessException when attempting to access gpio

NeK000, 2019-11-29:
**Describe the bug**
A clear and concise description of what the bug is.

I have developed a small Micro service for dht22 sensor. I am trying to initialize the                 Pi.Init<BootstrapWiringPi>();
as described in documentation and when I do that I am receiving back System.IO.IOException: Permission denied.

I have deployed the application using:
dotnet publish -r linux-arm -c Release

One of the endpoints is to activate the DHT sensor readings and as soon as I call it fails with the exception mentioned above from the stack trace I am seeing it goes throughout 
Pi.Init -> BootstrapWiringPi.Bootstrap -> EmbeddedResources.ExtractAll()

I have Pi 3 with Rasbian installed.

However Pi.info runs successfully

The application is developed netcore 3 version
nuget dependencies to 
Unosquare.Raspberry.Abstractions0.4.0
Unosquare.Raspberry.IO0.27.0
Unosquare.Raspberry.IO.Peripherals 0.5.0
Unosquare.WiringPi 0.5.0

NeK000, 2019-11-29:
I also tried changing the permission of the folder but none of it seems to help

k3z0, 2019-12-02:
Hi @NeK000 

The WiringPi bootstrap process extracts some files (WiringPi library) to your local folder and gives it execution permissions. This is necessary in order to use WiringPi functionality.

Since `Pi.info` doesn't fully depend on external libraries, like WiringPi, it can be used without bootstrapping WiringPi.

Try to run your app using `sudo` in order to give permissions to the app to extract the required files.

NeK000, 2019-12-02:
Hi @k3z0 

Thank you for the quick reply :)

I will try it out tmrw and see how it goes

For now I will close the issue if the suggested fix doesnot fix it I will reopen it and write again

closed 2019-12-02 by NeK000