unosquare/raspberryio #197: Debugging/Troubleshooting SPI device communication

manijak, 2019-12-23:
I'm trying to use this library in order to communicate with a Waveshare ePaper display. They have drivers and demos for c and python but not for .net. I've tested the provided demos, the display works as it should (using c & python). 

A fellow developer has already created a library in .net that runs his 2.7" ePaper display, and according to him, it does work. I have forked his repo in order to get my 7.5" ePaper display working, but so far I have not had any success. When running the commands, nothing happens on the display, no reaction whatsoever. I get no exceptions or errors during the run... 

My question is, is there a way I could debug or troubleshoot my code in order to find out why nothing is being displayed on my ePaper display? Is there a way to check if each command was successful or not?

My environment:
- Raspbian Buster
- .Net Core 3.1 
- Raspberry pi 3b+
- Installed both wiringpi and bcm drivers on the raspbian. 

My repo-fork can be found here: https://github.com/manijak/EPaperHatCore
The monitor in question: https://www.waveshare.com/product/modules/oleds-lcds/e-paper/7.5inch-e-paper-hat-c.htm

k3z0, 2019-12-26:
Hi @manijak,

Analyzing your code I realized that you are missing an actual implementation of RaspberryIO.Abstractions. You need to use something like [wiringpi-dotnet.
](https://github.com/unosquare/wiringpi-dotnet/)

Check the readme to see how to [install ](https://github.com/unosquare/raspberryio/#installation) and [use](https://github.com/unosquare/raspberryio/#usage) wiringpi-dotnet along with RaspberryIO.

manijak, 2019-12-26:
Oh yes I have that. The repo you are looking at is only the library for the drivers, I have separate project for running the code, it is not part of that repo.

My app has the following launch code:

```
static void Main(string[] args)
        {
            Pi.Init<BootstrapWiringPi>();
            Console.WriteLine(Pi.Info.BoardModel);

            //initialize ePaper display
            var ePaper = new Epaper_7in5bc(384, 640);
            ePaper.Initialize();

            ePaper.ClearScreen();

            //Thread.Sleep(500);
            Pi.Timing.SleepMilliseconds(500);
...
```

closed 2020-03-03 by stale[bot]