unosquare/raspberryio #198: RFID/NFC PN532 module device.DetectCard() does not work

LMSSonos, 2020-01-02:
**Describe the bug**
I have attached a PN532 module (https://www.elechouse.com/elechouse/index.php?main_page=product_info&cPath=90_93&products_id=2242) to the Raspberry Pi3 B+. The console application is netcore 3.1 with following code

```
            Pi.Init<BootstrapWiringPi>();
            Console.WriteLine("Testing RFID");
            while (true)
            {
                var device = new RFIDControllerMfrc522(Pi.Spi.Channel0, 280000, Pi.Gpio[22]);
                // If a card is found
                var result = device.DetectCard();
                if (result == RFIDControllerMfrc522.Status.Error) continue;
                Console.WriteLine("Card detected");

                // Get the UID of the card
                var uidResponse = device.ReadCardUniqueId();
            }
```

I expected to get the "Card detected" output, but the the result is always `RFIDControllerMfrc522.Status.Error` so it does seem able to read the NFC. I'm not sure about the Pi.GPio port I should pass, but I tried all from 18 to 24 and none helped.

**Wireing**
```
NFC  => Raspberry
SCK  => Pin 23
MISO => Pin 21
MOSI => Pin 19
SS   => Pin 24
VCC  => Pin 2
GND  => Pin 6
IRQ  => not connected
RSTO => not connected
```
**Devices**
Raspberry PI3 Model B+ with Raspian buster lite  v2019-09-26

**libnfc output**

nfc-list:
```
nfc-list uses libnfc libnfc-1.7.1-223-gf8b2852
NFC device: _PN532_SPI opened
```
nfc-poll
```
nfc-poll uses libnfc libnfc-1.7.1-223-gf8b2852
NFC reader: _PN532_SPI opened
NFC device will poll during 30000 ms (20 pollings of 300 ms for 5 modulations)
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  44
       UID (NFCID1): 04  70  9b  2a  26  5d  80
      SAK (SEL_RES): 00
Waiting for card removing...nfc_initiator_target_is_present: Target Released
done.
```

k3z0, 2020-01-02:
Hi @LMSSonos,

Well, I'm not sure if you can use a PN532 module with our MFRC522 module implementation since each card uses a different chip.

In order to test it, you need to ensure that your PN532 module is configured to use SPI as the communications protocol. Next, connect the _RSTO_ pin to the Raspberry's physical pin 15 (BCM22).

Try it out and let me know if it worked.

Thanks.

LMSSonos, 2020-01-02:
Hi @k3z0 

Thanks for fast reply. The chip is configured for SPI (see screenshot). I have also attached the RSTO to pin 15. `nfc-poll` still works, sample code from above does not. I tried with `Pi.Gpio[x]` (x tried 19 to 25).

[image]

k3z0, 2020-01-03:
Well, maybe the implementation does not support PN532 chip since it supposed to be used with RC522 chips.

LMSSonos, 2020-01-03:
Thank you. I did some research and found netcore code for the PN532 on github, but looks much more complex than the RC522 implementation made in this library.

So I will order an RC522 chip. With that I can read the contents of a MIFARE Ultralight NTAG213?

Again, thanks for all your help and time!

k3z0, 2020-01-03:
You need to read the [NTAG213 datasheet](https://www.nxp.com/docs/en/data-sheet/NTAG213_215_216.pdf) in order to know if it is compatible or how to use it with the RC522 chip.

LMSSonos, 2020-01-08:
@k3z0 I have no a RC522 reader. When I use the playground app, it detects the `MIFARE Ultralight C` tag with the card UID. If I use read card or read sector from the playground, I get following output

```
Testing RFID
Card UID: 136,4,123,334
Authentication error
```

In https://www.nxp.com/docs/en/data-sheet/MFRC522.pdf I found the following information

> 10.3.1.9    MFAuthent
> This command manages MIFARE authentication to enable a secure communication to any MIFARE Mini, MIFARE 1K and MIFARE 4K card.

But the specification also notes in the intro that 522 is compatile with MIFARE Ultralight. So I removed the ` device.AuthenticateCard1A` part in `Peripherals.Rfid.cs` file and now it works.

Should I make a PR with and extension of the Playground App where you can select Read without authentication?

Thanks for your help

k3z0, 2020-01-08:
@LMSSonos that sounds great, we are glad that you can contribute with this project.

LMSSonos, 2020-01-17:
Pull request created #200

closed 2020-01-17 by LMSSonos