unosquare/raspberryio #199: Unosquare GPIO pins not working with raspberry pi 4

scrumjunkie, 2020-01-16:
**Describe the bug**
Hi, I'm using unosquare to work with ultransonic hc-sr04 sensor. I've got it working with the raspberry 2 model b, but when I try exactly the same with raspberry pi model 4b it doesn't work. On the raspberry pi 4 the sensors always return -1. I've checked multiple times if the sensors were connected correctly. I've tested the sensors with some python code and with that code the sensor worked. After this I've also tried to make a led blink, but it seemed like the gpio pins didn't give any output. Are the GPIO's not working with unosquare on the pi model4b?

**The code I use**

```
using System;
using Unosquare.RaspberryIO;
using Unosquare.RaspberryIO.Peripherals;
using Unosquare.RaspberryIO.Abstractions;
using Unosquare.WiringPi;

namespace RaspberryIOTest3
{
    class Program
    {
        static void Main(string[] args)
        {
            Pi.Init<BootstrapWiringPi>();

            UltrasonicHcsr04 sensor1 = new UltrasonicHcsr04(Pi.Gpio[BcmPin.Gpio17], Pi.Gpio[BcmPin.Gpio27]);
            sensor1.OnDataAvailable += Sensor_OnDataAvailable;
            sensor1.Start();
        }

        private static void Sensor_OnDataAvailable(object sender, UltrasonicReadEventArgs eventArgs)
        {
            //eventArgs.Distance always is -1 on the raspberry pi 4
            Console.WriteLine("Distance: " + eventArgs.Distance);
        }
    }
}
```

**Expected behavior**
The proximity sensor returns a distance

**Devices**
Raspberry pi 4 info:
System Information
	LibraryVersion        : 2.50
	RaspberryPiVersion    : Pi4ModelB2Gb
	BoardRevision         : 2
	ProcessorCount        : 4
	InstalledRam          : 2033684480
	IsLittleEndian        : True
	ModelName             : ARMv7 Processor rev 3 (v7l)
	Features              : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
	CpuImplementer        : 0x41
	CpuArchitecture       : 7
	CpuVariant            : 0x0
	CpuPart               : 0xd08
	CpuRevision           : 3
	Hardware              : BCM2835
	Revision              : b03111
	RevisionNumber        : 1
	Serial                : 1000000000a87446
	UptimeTimeSpan        : 00:15:20.2500000

GPIO version 2.52

**Software versions**
Unosquare Raspberry.IO: 0.27.0
Unosquare.Raspberry.Abstractions: 0.4.0
Unosquare.Raspberry.Peripherals: 0.5.0
Unosquare.WiringPi: 0.5.0
Unosquare.Swan: 2.3.1
Unosquare.Swan.Lite: 2.3.0

k3z0, 2020-01-20:
Hi @scrumjunkie 

Unfortunately at this moment, we don't have a Raspberry 4 to perform tests. Maybe the problem is generated by the underlying library. Could you please try [updating WiringPi](http://wiringpi.com/wiringpi-updated-to-2-52-for-the-raspberry-pi-4b/) maybe this could help to solve the problem.

pgrawehr, 2020-01-24:
This may be a duplicate of #176

closed 2020-03-31 by stale[bot]