unosquare/raspberryio #203: Support for running in a container?

thnk2wn, 2020-01-28:
**Is your feature request related to a problem? Please describe.**
Currently it appears difficult to use raspberryio in a container (or at least not obvious how to).

I see this error

<pre>
wiringPiSetup: Unable to open /dev/mem or /dev/gpiomem: No such file or directory.
  Aborting your program because if it can not access the GPIO
  hardware then it most certianly won't work
  Try running with sudo?
</pre>

Currently using .Net Core 3.1 and:

```xml
 <PackageReference Include="Unosquare.Raspberry.IO" Version="0.27.0" />
 <PackageReference Include="Unosquare.WiringPi" Version="0.5.0" />
```
Dockerfille:

<pre>
# Stage 1
FROM mcr.microsoft.com/dotnet/core/sdk:3.1 AS build
ENV DOTNET_CLI_TELEMETRY_OPTOUT 1
WORKDIR /build
COPY . .
RUN dotnet restore
RUN dotnet publish -c Release -o /app -r linux-arm
# Stage 2
FROM mcr.microsoft.com/dotnet/core/aspnet:3.1.1-buster-slim-arm32v7 AS final
WORKDIR /app
COPY --from=build /app .
ENTRYPOINT ["dotnet", "./siren.dll"]
</pre>

Project: https://github.com/thnk2wn/rasp-cat-siren/tree/master/siren

**Describe the solution you'd like**
* Instructions that don't assume deployment is done via dotnet-sshdeploy
* Instructions for any changes to make running in a container both from app level and anything that needs to be setup as a prerequisite

**Describe alternatives you've considered**
I've looked at these threads so far and am still experimenting:
* https://github.com/unosquare/raspberryio/issues/62
* https://raspberrypi.stackexchange.com/questions/40105/access-gpio-pins-without-root-no-access-to-dev-mem-try-running-as-root

**Additional context**
Add any other context or screenshots about the feature request here.

logicethos, 2020-01-28:
Does this help? https://stackoverflow.com/questions/30059784/docker-access-to-raspberry-pi-gpio-pins

thnk2wn, 2020-02-17:
Yeah, thanks.

closed 2020-02-17 by thnk2wn