unosquare/raspberryio #204: WiringPi does only support a simple interrupt callback that has no parameters

logicethos, 2020-01-29:
**Describe the bug**

After I do 
`Pi.Init<BootstrapWiringPi>(); `
I get this error
`WiringPi does only support a simple interrupt callback that has no parameters.`

**To Reproduce**
I copied /usr/lib/libwiringPi.so.2.50 to my app directory.
I had to do this to overcome this issue https://github.com/unosquare/raspberryio/issues/62

**Devices**
Pi 2 Model B
Raspbian GNU/Linux 9.8 (stretch)

k3z0, 2020-01-30:
Hi @logicethos ,

Actually, that message is not an error, That message is indicating that WiringPi does not support a callback with parameters as interrupt callback, you must call the overload with the Action that does not take any parameter: `RegisterInterruptCallback(EdgeDetection edgeDetection, Action callback)`.

logicethos, 2020-01-30:
Ah, understood. Thank you.

closed 2020-01-30 by logicethos