unosquare/raspberryio #209: Pi.Camera.CaptureImage(settings) returns empty array when running from container

pegouk, 2020-04-02:
**Describe the bug**
Using the code below, Pi.Camera.CaptureImage(settings) returns an empty array when run from a container. No errors are generated.
When run from command line `dotnet TestApp.dll`, runs successfully. Returns an array of length 10510810.

`            
Pi.Init<BootstrapWiringPi>();
camSettings = new Unosquare.RaspberryIO.Camera.CameraStillSettings
{
   CaptureExposure = Unosquare.RaspberryIO.Camera.CameraExposureMode.Auto,
   CaptureDynamicRangeCompensation = Unosquare.RaspberryIO.Camera.CameraDynamicRangeCompensation.Off,
   CaptureEncoding = Unosquare.RaspberryIO.Camera.CameraImageEncodingFormat.Jpg,
   CaptureMeteringMode = Unosquare.RaspberryIO.Camera.CameraMeteringMode.Average,
   CaptureWhiteBalanceControl = Unosquare.RaspberryIO.Camera.CameraWhiteBalanceMode.Fluorescent,
   CaptureJpegIncludeRawBayerMetadata = true,
   CaptureJpegQuality=100
};

//Camera "warm-up" time
Thread.Sleep(5000);
byte[] pic = null;
pic = Pi.Camera.CaptureImage(camSettings);
`

**Expected behavior**
An array of length 10510810, similarly to array returned when running from command line .

**Devices**
    Unosquare.Raspberry.IO Version="0.27.0"
    Unosquare.WiringPi Version="0.5.0"
    3.1.2-buster-slim-arm32v7 docker image (running privileged)
    OS Raspbian GNU/Linux 10 (buster)
    Raspberry Pi 3+
    Pi Camera v2

geoperez, 2020-04-02:
This method is just a wrapper over `raspistill`, are you able to run this command from a container?

pegouk, 2020-04-15:
Yes, I can run raspistill with no problems

crazybg321, 2020-05-03:
@pegouk this is not a bug in the library. I had the same issue. Check this answer:

https://github.com/unosquare/raspberryio/issues/208

In addition to what he did, you should add the `/opt/vc/lib/` to the library path of your container and then it will work. Basically include this in your `Dockerfile`: `ENV LD_LIBRARY_PATH="/opt/vc/lib:${LD_LIBRARY_PATH}"`

@geoperez would it be an option for someone to add a manual for creating Docker containers? I found it quite tricky too to figure it out in the blind. I could help if you want.

pegouk, 2020-05-04:
Thanks, that solved it. I agree that it is quite tricky to figure out the Docker containers. I tried a number of things but not this one. Would be great to have some sort of manual for this.

closed 2020-05-04 by pegouk